rwth-afu/UniPager #3: Bei mehreren Aussendungen hintereinander kommt beim Alphapoc nur Kauderwelsch an

rtausend, 2017-01-09:
Hallo zusammen,

wenn ich eine Testaussendung mit 10 Nachrichten mache kommt bei einem Alphapoc Empfänger nur noch Kauderwelsch an. Anbei die Fotos von den einzelnen Nachrichten. Ich habe die Vermutung das die Nachrichten zu schnell hintereinander ankommen für den Empfänger. Vielleicht kann man da am Timing was drehen? Mit dem Skyper scheint dieses Problem nicht aufzutreten.

INFO - Received Message { id: 19, mtype: AlphaNum, speed: Baud(1200), addr: 399881, func: AlphaNum, data: "No. 0 Long bulk paging test for the RasPagerC9000, AFU RWTH Aachen 019" }
INFO - Waiting for TimeSlot(2)...
INFO - Transmitting...
INFO - Sending data...
Wiedergabe: Rohdaten 'stdin' : Unsigned 8 bit, Rate: 48000 Hz, mono
INFO - Data sent.
INFO - Transmission completed.
INFO - Received Message { id: 20, mtype: AlphaNum, speed: Baud(1200), addr: 399881, func: AlphaNum, data: "No. 1 Long bulk paging test for the RasPagerC9000, AFU RWTH Aachen 019" }
INFO - Waiting for TimeSlot(2)...
INFO - Transmitting...
INFO - Sending data...
Wiedergabe: Rohdaten 'stdin' : Unsigned 8 bit, Rate: 48000 Hz, mono
INFO - Received Message { id: 21, mtype: AlphaNum, speed: Baud(1200), addr: 399881, func: AlphaNum, data: "No. 2 Long bulk paging test for the RasPagerC9000, AFU RWTH Aachen 019" }
INFO - Received Message { id: 22, mtype: AlphaNum, speed: Baud(1200), addr: 399881, func: AlphaNum, data: "No. 3 Long bulk paging test for the RasPagerC9000, AFU RWTH Aachen 019" }
INFO - Received Message { id: 23, mtype: AlphaNum, speed: Baud(1200), addr: 399881, func: AlphaNum, data: "No. 4 Long bulk paging test for the RasPagerC9000, AFU RWTH Aachen 019" }
INFO - Received Message { id: 24, mtype: AlphaNum, speed: Baud(1200), addr: 399881, func: AlphaNum, data: "No. 5 Long bulk paging test for the RasPagerC9000, AFU RWTH Aachen 019" }
INFO - Received Message { id: 25, mtype: AlphaNum, speed: Baud(1200), addr: 399881, func: AlphaNum, data: "No. 6 Long bulk paging test for the RasPagerC9000, AFU RWTH Aachen 019" }
INFO - Received Message { id: 26, mtype: AlphaNum, speed: Baud(1200), addr: 399881, func: AlphaNum, data: "No. 7 Long bulk paging test for the RasPagerC9000, AFU RWTH Aachen 019" }
INFO - Received Message { id: 27, mtype: AlphaNum, speed: Baud(1200), addr: 399881, func: AlphaNum, data: "No. 8 Long bulk paging test for the RasPagerC9000, AFU RWTH Aachen 019" }
INFO - Data sent.
INFO - Transmission completed.
INFO - Waiting for TimeSlot(2)...
INFO - Transmitting...
INFO - Sending data...
Wiedergabe: Rohdaten 'stdin' : Unsigned 8 bit, Rate: 48000 Hz, mono
INFO - Received Message { id: 28, mtype: AlphaNum, speed: Baud(1200), addr: 399881, func: AlphaNum, data: "No. 9 Long bulk paging test for the RasPagerC9000, AFU RWTH Aachen 019" }
INFO - Data sent.
INFO - Transmission completed.

[bilder.zip](https://github.com/rwth-afu/RustPager/files/693309/bilder.zip)

7h0ma5, 2017-01-09:
Vielen Dank für die Problemberichte! Es kann sein, dass dieses Problem mit #2 zusammenhängt. Anscheinend gibt es Sonderfälle, in denen die POCSAG-Generierung noch nicht richtig funktioniert. Ich werde mir das nochmal genauer ansehen.

rtausend, 2017-01-09:
Hi! Das Problem gibt es mit der alten Java Version auch schon, wenn du jemand zum testen brauchst, sag Bescheid, kann das mit meinem Pager immer nachstellen 

Gruß Robert