rwth-afu/UniPager #6: Nagios Ausgabe, ob Master connected ist

dh3wr, 2017-01-09:
Am schönsten natürlich eine Nagios-kompatibel formatierte Ausgabe in einer Datei (Ort siehe #5 )

MasterConnection OK - Connected to <IP>:<Port>   oder
MasterConnection CRITICAL - Not connected

rtausend, 2017-01-10:
Für Zabbix wäre das natürlich auch genial :)

dh3wr, 2017-01-10:
@rtausend : Hast du ein Beispiel, wie die Ausgabe aussehen muss?

rtausend, 2017-01-10:
ich hab gerade geschaut, dafür gibts wohl keinen Standard. Ich nehm was ich bekomm und parse mir das dann. Vielen Dank

dh3wr, 2017-03-13:
Also da verwenden wir dann wohl auch einfach das Websocket. Client wäre hier https://blog.grandcentrix.net/a-command-line-websocket-client/
@7h0ma5 sind die Websocket-Befehle irgendwo dokumentiert? Vielleicht im Wiki des Github-Repos?

7h0ma5, 2017-03-13:
Ich habe mal damit angefangen, ist aber noch unvollständig: https://github.com/rwth-afu/RustPager/wiki/Websocket-Protocol

dh3wr, 2017-03-17:
@7h0ma5 Danke für die Infos. Ich habe unter https://github.com/rwth-afu/RustPagerNagios ein passenden Plugin geschrieben. Info dazu unter https://www.afu.rwth-aachen.de/projekte/funkruf-pager-pocsag/rustpager#nagios
Damit können wir das schließen.

closed 2017-03-17 by dh3wr