rwth-afu/UniPager #10: Txdelay für Audio-Variante

dh3wr, 2017-01-10:
Manche Sender brauchen ein TX-Delay. Die Zeit beim Zusammenbau der Aussendung von der insg. zur Verfügung stehende Zeit abgezogen werden, da sie ja Teil der Aussendung ist und nicht die ganze Sendung um den TX-Delay-Wert nach hinten verschoben werden soll. Sonst hängt das Ende der Sendung schon im nächsten, nicht-mehr-erlaubten-Zeitschlitz.

7h0ma5, 2017-01-10:
Die Audio-Ausgabe funktioniert momentan noch so, dass der komplette Datenstrom vor dem Abspielen generiert wird und dann an aplay gepiped wird. Da es sich um vom Compiler optimierten Maschinencode handelt, ist das Delay dadurch relativ klein. Ziel ist es aber, das Audiosignal parallel zum Abspielen zu generieren, damit neue Nachrichten, die während der aktuellen Sendung ankommen, auch noch mit eingebaut werden können, so wie das beim C9000 und Raspager auch schon funktioniert. Der Scheduler wird noch erweitert, so dass die genau benötigte Zeit inklusive TX-Delay berechnet wird.

dh3wr, 2017-01-10:
Sehr gut. Die PTT-Leitung muss dann natürlich ganz am Anfang schon aktiviert werden.

commit 5448a37b (2017-01-12): Implement TX delay for AudioTransmitter (close #10)
closed 2017-01-12 by 7h0ma5