rwth-afu/UniPager #11: Alsa-Device Auswahl

dh3wr, 2017-01-10:
Nicht immer ist die Default-Karte die richtige..

7h0ma5, 2017-01-10:
Spricht etwas dagegen, die Default-Karte zu ändern? Ein Text-Feld zur Eingabe des Devices in der Art "hw:0,0" wäre einfach, eine Liste wesentlich komplizierter.

dh3wr, 2017-01-10:
Vielleicht kann man mit http://diwic.github.io/alsa-rs-docs/alsa/device_name/index.html die vorhandenen Devices einlesen? Sonst über ein geparste Ausgaben von aplay -L oder aplay -l ?  Eine Dropdown-Liste wäre schon schöner...

dh3wr, 2017-03-02:
Ich habe nun 2 Anfragen zu diesem Thema. Könnte man nicht auf die schnelle ein Textfeld in die Weboberfläche einbauen, wo man den Soundcard-Identifier als Klartext eingeben kann und dieser wird dann hier
https://github.com/rwth-afu/RustPager/blob/master/src/transmitter/audio/transmitter.rs Line 68 ff. :
````
       let mut child = Command::new("aplay")
            .stdin(Stdio::piped())
            .args(&["-t", "raw", "-N", "-f", "U8", "-c", "1"])
            .args(&["-r", &*format!("{}", SAMPLE_RATE)])
            .spawn()
            .expect("Failed to start aplay");
````
eingabaut?

closed 2017-03-05 by 7h0ma5

dg1kwa, 2017-03-15:
Frage: Was MUSS man in diesem Feld den jetzt eingeben? Bei gar nicht zieht default?

stephensworld, 2017-05-11:
das würde mich auch interessieren.

Grüße
Stephan

dh3wr, 2017-05-11:
Dazu kann sich am besten @7h0ma5 äußern.

7h0ma5, 2017-05-12:
Wenn nichts in das Feld eingegeben wird, wird das `default`-Device verwendet. Wenn man eine bestimmte Soundkarte verwenden will, muss man das im Format `plughw:[device id],[subdevice id]` angeben, zum Beispiel `plughw:1,0`.

commit 5b0891d6 (2017-03-05): Implement ALSA device setting (close #11)