rwth-afu/UniPager #7: Aussendung bleibt einfach hängen

rtausend, 2017-01-10:
Hallo,

mir ist gestern schon aufgefallen das nach einiger Laufzeit die Aussendungen einfach nicht mehr statt finden. Hatte den Server gestern dann neu gestartet, heute morgen aber das gleiche spiel.

Im log sieht man nur folgendes:

Connected to RustPager.

INFO Received Message { id: 0, mtype: AlphaNum, speed: Baud(1200), addr: 399881, func: AlphaNum, data: "test123" }

INFO Received Message { id: 68, mtype: Numeric, speed: Baud(1200), addr: 2504, func: Numeric, data: "083200 100117" }

INFO Received Message { id: 69, mtype: Numeric, speed: Baud(1200), addr: 2504, func: Numeric, data: "083400 100117" }

Es findet aber keine Aussendung mehr statt bis man den ruspager neu startet.

Ich weis jetzt nicht ob das mit der Änderung von heute Nacht behoben wurde, ich hatte gegen 23Uhr nochmal ne neue Version gebaut (0.3.1)

Danke und Gruß

rtausend, 2017-01-10:
Im Logfile habe ich noch diese Meldungen hier gefunden, ich hab aber keine Ahnung ob das was damit zu tun hat. Mir ist nur aufgefallen das als ich den rustpager neu gestartet hatte, die Meldung nochmal kam, danach dann aber erst mal nicht mehr:

Jan 10 08:29:15 raspberry kernel: [34150.746701] w1_master_driver w1_bus_master1: Family 0 for 00.034000000000.79 is not registered.
Jan 10 08:30:00 raspberry rsyslogd-2007: action 'action 17' suspended, next retry is Tue Jan 10 08:31:30 2017 [try http://www.rsyslog.com/e/2007 ]
Jan 10 08:30:03 raspberry kernel: [34199.087981] w1_master_driver w1_bus_master1: Family 0 for 00.834000000000.f5 is not registered.
Jan 10 08:31:03 raspberry kernel: [34259.267146] w1_master_driver w1_bus_master1: Family 0 for 00.434000000000.3f is not registered.
Jan 10 08:31:34 raspberry rsyslogd-2007: action 'action 17' suspended, next retry is Tue Jan 10 08:33:04 2017 [try http://www.rsyslog.com/e/2007 ]
Jan 10 08:31:50 raspberry kernel: [34305.678570] w1_master_driver w1_bus_master1: Family 0 for 00.c34000000000.b3 is not registered.
Jan 10 08:32:14 raspberry kernel: [34329.987546] w1_master_driver w1_bus_master1: Family 0 for 00.234000000000.5a is not registered.
Jan 10 08:33:02 raspberry kernel: [34378.317724] w1_master_driver w1_bus_master1: Family 0 for 00.a34000000000.d6 is not registered.
Jan 10 08:33:20 raspberry rsyslogd-2007: action 'action 17' suspended, next retry is Tue Jan 10 08:34:50 2017 [try http://www.rsyslog.com/e/2007 ]
Jan 10 08:33:39 raspberry kernel: [34414.577951] w1_master_driver w1_bus_master1: Family 0 for 00.634000000000.1c is not registered.
Jan 10 08:34:25 raspberry kernel: [34461.048139] w1_master_driver w1_bus_master1: Family 0 for 00.e34000000000.90 is not registered.
Jan 10 08:35:01 raspberry rsyslogd-2007: action 'action 17' suspended, next retry is Tue Jan 10 08:36:31 2017 [try http://www.rsyslog.com/e/2007 ]
Jan 10 08:35:14 raspberry kernel: [34509.788672] w1_master_driver w1_bus_master1: Family 0 for 00.134000000000.e4 is not registered.
Jan 10 08:36:01 raspberry kernel: [34557.099086] w1_master_driver w1_bus_master1: Family 0 for 00.934000000000.68 is not registered.
Jan 10 08:37:01 raspberry kernel: [34616.389269] w1_master_driver w1_bus_master1: Family 0 for 00.534000000000.a2 is not registered.
Jan 10 08:37:01 raspberry rsyslogd-2007: action 'action 17' suspended, next retry is Tue Jan 10 08:38:31 2017 [try http://www.rsyslog.com/e/2007 ]
Jan 10 08:38:00 raspberry kernel: [34675.669538] w1_master_driver w1_bus_master1: Family 0 for 00.d34000000000.2e is not registered.
Jan 10 08:38:59 raspberry rsyslogd-2007: action 'action 17' suspended, next retry is Tue Jan 10 08:40:29 2017 [try http://www.rsyslog.com/e/2007 ]
Jan 10 08:39:11 raspberry kernel: [34746.880088] w1_master_driver w1_bus_master1: Family 0 for 00.334000000000.c7 is not registered.

rtausend, 2017-01-10:
Ich musste tatsächlich nochmal den ganzen Raspberry neu starten um ihn wieder zum senden zu überraden ... das ist komisch

dh3wr, 2017-01-10:
Das sieht nach OneWire Bus aus. Was hat das denn mit dem Rustpager zu tun?

rtausend, 2017-01-10:
DAS kann ich natürlich nicht sagen, ich war mal so frei und hab den einfach abgeschaltet und den Raspberry rebootet. Ich beobachte weiter

7h0ma5, 2017-01-10:
Dieses Problem tritt auf, wenn der Scheduler-Thread abgestürzt ist, da die Nachrichten dann nicht mehr verarbeitet werden. Version 0.3.1 hatte noch einen Bug, der öfters zum Absturz des Scheduler-Threads geführt hat, dieser Bug sollte mit Version 0.3.2 bereits behoben sein. Zusätzlich werde ich die Ausgabe einer Fehlermeldung implementieren, falls der Scheduler-Thread abgestürzt sein sollte, vielleicht auch einen automatischen Neustart, aber eigentlich sollte es gar nicht erst so weit kommen.

7h0ma5, 2017-01-12:
Seit Version 0.3.2 konnte ich das Problem nicht mehr reproduzieren, daher gehe ich erstmal davon aus, dass es behoben ist.

closed 2017-01-12 by 7h0ma5