rwth-afu/UniPager #8: Logfile

rtausend, 2017-01-10:
Hallo,

es wäre super wenn man irgendwie ein Logfile ausgeben könnte z.B. nach /var/log/rustpager.log (oder konfigurierbar)

Im moment könnte man einen Fehler der nachts passiert glaube ich nicht nachvollziehen oder?

Danke und Gruß

7h0ma5, 2017-01-10:
Es wäre kein Problem das zu Implementieren, aber da das Log auch auf der Standardausgabe ausgegeben wird, könnte man die komplette Ausgabe auch mit dem Befehl "./RustPager &>> rustpager.log" in die Datei rustpager.log umleiten. Wenn man systemd benutzt, dann geht es auch so schon, da kann man sich das Log einfach mit "sudo journalctl -u rustpager" anzeigen lassen.

rtausend, 2017-01-10:
Verstehe das Argument, nur zum "mal schnell" durchsuchen dürfte ein echtes Logfile doch praktischer sein oder bin ich nur zu "old scool" :) Ich hab das bei mir jetzt so implementiert das er über den systemd gestartet wird, funktioniert auch sehr gut

daccle, 2017-02-28:
Hmm, @7h0ma5 hat gezeigt, wie es mit systemd/jounalctl geht. IMHO kann das hier geschlossen werden.

closed 2017-03-14 by 7h0ma5