rwth-afu/UniPager #12: Username / Passwort für die Settings auf der Website

rtausend, 2017-01-11:
Hallo,

ich glaube es wäre sicher sinnvoll noch eine Username/Passwortabfrage für die Einstellungen zu implementieren. Sicher im AFU soll alles offen sein, aber ich denke ein Senderbetreiber möchte nicht das jeder den Sender verstellen kann.

Vielleicht ist dieses Erweiterung auch schon angedacht, wollte es nur mal angesprochen haben.

Gruß Robert

daccle, 2017-02-28:
@rtausend vielleicht hilft Dir das iptables Beispiel in https://github.com/rwth-afu/RustPager/issues/37 ? Dafür muss natürlich die default policy drop sein. So kann man das Webinterface zumindest nur von lokal aus sehen.

rtausend, 2017-03-24:
Das ist ne gute idee, aber vielleicht möchte ich die Website ja auch von anderen Rechner erreichen, ich denke ein Passwortschutz ist hier schon wichtig

daccle, 2017-03-24:
Dafür gibt es ssh-tunnel oder richte halt einen Proxy mit einem Webserver Deiner Wahl ein. (Ich verstehe Deinen Featurewunsch trotzdem ;-) )

dh3wr, 2017-05-26:
Ich würde auch sagen: Was wäre die nächste Aktion in der Prio-Liste. Insb. da nun die Testbutton-Fkt. eingebaut ist.

dh3wr, 2017-12-08:
Hier würde ich auch die Prio etwas hoch setzen :-)

7h0ma5, 2017-12-28:
Man kann jetzt ein Passwort setzen, indem man die Datei `/var/lib/unipager/password` anlegt und das Passwort da rein schreibt. Danach muss man den UniPager einmal neustarten. Ich werde bald auch noch eine Option zum Setzen des Passworts in das Webinterface einbauen, das ist aber etwas komplizierter.

closed 2017-12-28 by 7h0ma5

DL1AFA, 2018-03-10:
Muss man bei der Vergabe des Passwortes irgendetwas besonderes beachten? Bei mir funktioniert das nicht oder habe ich das falsch verstanden, dass das Passwort dann für die Webschnittstelle gilt?

dh3wr, 2018-03-12:
Ja das PW gilt dann für das Web-Interface auf Port 8073 und intern auch für Websocket auf Port 8055. Ziel ist es, dass nicht jeder die Unipager-Einstellungen ändern kann.

M6NBP, 2019-04-24:
I can not log in via a 2nd PC 
Ports 8073 and 8055 is open 
I have made /var/lib/unipager/password and entered password 

I still can not get pass the password

M6NBP, 2019-04-24:
Found the issue
It is the Adaware Antivirus Pro
No way to set a rule to over ride.
So for now the only way is to turn it off

commit d44b2fca (2017-12-28): Implement authentication (close #12)