rwth-afu/UniPager #13: Log-Fenster Hold

dk4pa, 2017-01-13:
Bei eingehenden Nachrichten springt das Fenster in die letzte Zeile. 
Wenn man gerade in dem Log hochscrollt, um etwas nachzusehen, muss man jedes mal wieder hochscrollen.
Wäre hier eine Fuktion möglich ähnlich InvertPtt Button, ein Button Springe an Listenende bei neuen Nachrichten?
So könnte man kurzzeitig im Log scollen ohne jedesmal an das Listenende zu springen.
vy37, Stefan DK4PA

daccle, 2017-02-28:
Siehe den Kommentar zu journalctl in #8 Mit journalctl ist es eh praktischer und Du hast die Funktionalität sofort, statt auf das Enhancement zu warten. ;-)

dk4pa, 2017-03-06:
Ja funktioniert super, danke für den Hinweis. 
Schön wäre, wenn man sich von einem anderen Rechner einloggt, der Log seit dem letzten Start geladen wird.