rwth-afu/UniPager #15: Lange Nachrichten werden beim C9000 manchmal abgeschnitten

dj7lc, 2017-01-14:
...aber nicht immer. Ich vermute einen Pufferüberlauf.

7h0ma5, 2017-01-15:
Lässt sich das Problem reproduzieren, also tritt es immer bei der selben Pager-Adresse mit dem selben Text oder einem Text mit der gleichen Länge auf? Wenn ja, dann wäre es hilfreich, wenn du diese Fälle hier angeben könntest.

Es gibt einen Sonderfall im POCSAG-Generator, der momentan noch nicht berücksichtigt wird. Wenn eine Nachricht genau im letzten Codewort des aktuellen Batches endet und auch keine Nullen mehr zum Abschluss der Nachricht in das letzte Codewort passen, dann muss noch ein weiterer leerer Batch gesendet werden, damit der Pager das Ende der Nachricht korrekt erkennt, sonst gehen alle Buchstaben aus dem letzten Batch verloren. Ich vermute, dass du auf genau diesen Sonderfall gestoßen bist.

PS: Pufferüberläufe sind durch die Architektur von Rust nicht möglich :wink:

dj7lc, 2017-01-16:
Ich kann den Fehler reproduzieren, dazu verwende ich folgende Nachricht:
"LONGBULKPAGINGTEST1LONGBULKPAGINGTEST2LONGBULKPAGINGTEST3LONGBULKPAGINGTEST4LONG"
Wenn der Pager oder die Software-Decoder soRFmon und monitord überhaupt auswerten, sieht das Ergebnis immer wie folgt aus:
"LONGBULKPAGINGTEST1LONGBULKPAGINGTEST2LONGBULKPAGINGTES" + "ein paar Zufallszeichen"

Kürzere Nachrichten, wie ""LONGBULKPAGINGTEST1LONGBULKPAGINGTEST2LONGBULKPAGINGTES" werden, wenn ausgewertet, vollständig und korrekt angezeigt.

Noch was: Der Sendervorlauf (also der konstante Ton vor der Aussendung) klingt für mich sowohl beim SDRPager und dem C9000 akustisch viel zu kurz und ist manchmal auch gar nicht vorhanden. Das kann der Grund sein, warum der Pager und die Software-Decoder nicht immer auswerten. Meine Vermutung ist eine Verzögerung bei der PTT-Schaltung.

dj7lc, 2017-01-24:
Das Problem ist möglicherweise auf eine fehlerhafte Implementierung im Mikroprogramm des Steuerprozessors zurückzuführen. Die Angelegenheit wird geprüft :-)

dj7lc, 2017-02-26:
Auch im neuen Branch besteht der Fehler weiterhin, der Sender hört nach etwa dem 55. Zeichen einfach auf zu senden. Wenn mehrere kurze Funkrufe hintereinander abgesetzt werden, fällt der Sender auch nach dem insgesamt 55. Zeichen ab.

dh3wr, 2017-03-14:
Ich kann das auch in der Version v0.4.1 weiterhin bestätigen. Vielleicht eine Timing-Geschichte? Mit SDR# und POC32 das gleiche Bild. Es kommt also schon schlecht aus dem C9000 raus. Könntest du mir eine Raspi1 Version von v0.2.2 bauen? Mit der gings damals noch meine ich

dh3wr, 2017-03-14:
Ich bin mir jetzt doch nicht mehr sicher, ob es früher mal ging. Ich kann das aber auch im neuen C9000 mit Pi3 ausprobieren.

dj7lc, 2017-03-14:
Mit dem Java-Programm hatte es so weit ich weiß mal funktioniert. Das Problem dürfte in der PTT-Steuerung liegen.

dh3wr, 2017-03-14:
@dj7lc Das denke ich nicht: Hier https://github.com/rwth-afu/RasPagerC9000/blob/master/software/atmega/main.cpp sieht alles PTT-mäßig gut aus. Ich probiere mal mit dem Java-Programm, um den ATMega ausschließen zu können.

dh3wr, 2017-03-14:
Java Programm geht nicht, weil es irgendwie nicht mit dem Proxy will. Kann ich erst später nach schauen

dh3wr, 2017-03-14:
@7h0ma5 Kannst du mir vielleicht eine Test Version für einen Pi1 kompilieren, welche am Ende des Rufs noch 10 Mal ein-null macht? Dann kann ich mit einem RTL-SDR schauen, ob der Ruf komplett rausgeht, oder der Sender zu früh Schluss macht?

7h0ma5, 2017-03-14:
Ich habe mir sowohl den AVR-Code als auch den Transmitter-Code angesehen und eigentlich sieht da alles in Ordnung aus. Ich vermute eher, dass das Problem im POCSAG-Generator liegt. Dann müsste das Problem aber auch bei anderen Transmitter-Typen auftreten. Hat das mal jemand getestet?

dh3wr, 2017-03-14:
Ich habe jetzt bei DB0SDA C9000 das Java-Programm mit Proxy laufen. Geht jetzt. Lange Rufe gehen auch. Es muss also am RustPager liegen. Ob es bei anderen Sender-Arten auch ist, versucht ich mal durch eine Wiederbelegung des RasPagersV1 herauszufinden.

dh3wr, 2017-03-14:
Ich vergleiche gleich mal das Audio-Signal zwischen RustPager und Java-Pager. Der DB0SDA läuft ja jetzt auf Java, und den DB0WA C9000 lasse ich dann auf Rustpager laufen.

7h0ma5, 2017-03-14:
Ich habe gerade zwei mögliche Fehlerursachen behoben. Bitte jetzt nochmal testen. 
[rustpager-rpi.tar.gz](https://github.com/rwth-afu/RustPager/files/842252/rustpager-rpi.tar.gz)

dh3wr, 2017-03-14:
Schon deutlich besser. Aber jetzt werden am Ende noch 3 Zeichen angehängt (kryptisch) und teilweise sind die Rufe auch ganz kaputt.

dh3wr, 2017-03-14:
@7h0ma5 Ich habe dir mal einen dl2ic_debug Sender angelegt, weil du mir jede Sekunde Fehlermeldungen im Core gemacht hast

dh3wr, 2017-03-14:
Mal hier der Mitschnitt vom POC32:
[image]

dh3wr, 2017-03-14:
In Anbetracht, dass ich gerade 
````
print "curl -H \"Content-Type: application/json\" -X POST -u dh3wr:de1rcw -d '{ \"text\": \"No. ";
print $i;
print " Long bulk paging test for the RasPagerC9000, AFU RWTH Aachen ";
print $sec;
print $min;
print "\",\"callSignNames\": [\"dh3wr\",\"db6ac\",\"dl5fmk\"], \"transmitterGroupNames\": [\"all\"], \"emergency\": false }' http://dapnet.db0sda.ampr.org:8080/calls\n";
```` gefunden habe, scheint es mit dem Java Ding wohl gegangen zu haben.

7h0ma5, 2017-03-14:
@dh3wr Den dl2ic_debug Sender hatte ich eben angelegt, kurz bevor es Probleme mit dem Core gab. Wenn der nicht mehr da war nach dem Crash, dann sind wohl Daten verloren gegangen.

> Schon deutlich besser. Aber jetzt werden am Ende noch 3 Zeichen angehängt (kryptisch) und teilweise sind die Rufe auch ganz kaputt.

Ist dieses Problem wieder nur auf den C9000 beschränkt?

dh3wr, 2017-03-14:
Ich habe den Core auf einen alten Commit zurückgestellt, mit dem zumindest die Rufe wieder gehen. Ich mach mal DB0EIF an und schau damit.

dh3wr, 2017-03-14:
Das ist der Output mit RasPagerV1
[image]

Auch nicht ganz ok, aber immerhin besser

7h0ma5, 2017-03-14:
Liegt es vielleicht am TX Delay? Ich sehe gerade, dass in der Java-Version das TX Delay auf 10ms gesetzt ist, beim RustPager ist das nur auf 1ms.

dh3wr, 2017-03-14:
könnte sein. Wenn du magst, mach mir eine Version für Pi1 (db0sda) und Pi3B (db0wa)

dh3wr, 2017-03-14:
Vielleicht mehr als 1 ms warten auf die Antwort vom ATMega?

7h0ma5, 2017-03-14:
Das ist wohl das Problem. Der Send-Pin wird schon abgefragt, obwohl der Puffer des seriellen Ports vielleicht noch gar nicht vollständig geschrieben wurde. Der Flush-Befehl sollte eigentlich blockieren, bis alles geschrieben ist. Ich habe danach jetzt noch eine Millisekunde Delay eingefügt, bevor der Send-Pin das erste mal abgefragt wird. Wenn der Flush-Befehl nicht blockierend ist, dann habe ich momentan keine Idee, wie man das lösen könnte. Hier ist eine Binary die auf allen PIs laufen sollte: 
[rustpager-rpi.tar.gz](https://github.com/rwth-afu/RustPager/files/842690/rustpager-rpi.tar.gz)

7h0ma5, 2017-03-14:
In der Java-Version wird aus diesem Grund wahrscheinlich auch 10ms gewartet, wenn die 40 Bytes gesendet wurden. Wenn es noch nicht funktioniert, könnte ich das auch noch einbauen.

dh3wr, 2017-03-14:
Geht noch nicht, aber die Zeitschlitz-Sache ist ja schön... Wenngleich die Kiste schon auf Sendung war, auch wenn noch der letzte nicht-aktive Schlitz geblinkt hat

dh3wr, 2017-03-14:
40 Byte bis 1,2 kBaud brauchen ja min. 33 ms. Gut, da is 40 ms schon knapp....

dh3wr, 2017-03-15:
Ich wäre bereit für einen Test mit 10 ms delay eingebaut :)

7h0ma5, 2017-03-15:
Ich konnte leider gerade erst wieder an den Rechner, auf dem der Crosscompiler installiert ist. Hier ist die Version mit 10 ms Delay:  [rustpager-rpi.tar.gz](https://github.com/rwth-afu/RustPager/files/845451/rustpager-rpi.tar.gz)

dh3wr, 2017-03-16:
Interessantes systematisches Verhalten mit dieser Version: Der 1. Ruf ist ok, danach steigt die Anzahl der empfangenen Zeichen linear :-)
[image]

dh3wr, 2017-03-16:
Ich habe mit mal https://github.com/rwth-afu/RustPager/blob/master/src/transmitter/c9000/transmitter.rs
Wenn ich das richtig sehe, wir in Zeile 77 mit `` if (*self.serial).write_all(&bytes).is_err() {`` je gesendet. Und direkt im nächsten For-Durchlauf wird gefragt in Zeile 67
``while !self.send_pin.read() {``

D.h. zwischen Flush und read() sind nur ein paar Code-Zeilen. Kann der atmega da schon drauf geantwortet haben? Sollten wir vielleicht hier die 10 ms nach dem Flush einbauen? Oder ist das schon so?

7h0ma5, 2017-03-16:
In der aktuellen Testversion ist das schon so, vor jeder Abfrage des Pins wird 10 ms gewartet:

```rust
fn send(&mut self, gen: Generator) {
    self.ptt_pin.set_high();

    for (i, word) in gen.enumerate() {
        if i % 40 == 0 {
            if (*self.serial).flush().is_err() {
                error!("Unable to flush serial port");
            }

            thread::sleep(time::Duration::from_millis(10));
            while !self.send_pin.read() {
                thread::sleep(time::Duration::from_millis(1));
            }
        }

        let bytes = [((word & 0xff000000) >> 24) as u8,
                     ((word & 0x00ff0000) >> 16) as u8,
                     ((word & 0x0000ff00) >> 8) as u8,
                     ((word & 0x000000ff)) as u8];

        if (*self.serial).write_all(&bytes).is_err() {
            error!("Unable to write data to the serial port");
            self.ptt_pin.set_low();
            return;
        }
    }

    if (*self.serial).flush().is_err() {
        error!("Unable to flush serial port");
    }

    self.ptt_pin.set_low();
}
```

dh3wr, 2017-03-16:
Ich empfehle gleich mal den 8 Kanal logic analyser anzuschließen

commit 535c5f40 (2017-03-16): Fix C9000 communication bug (close #15)
closed 2017-03-16 by 7h0ma5