rwth-afu/UniPager #16: Automatische Rufzeichenaussendung

dj7lc, 2017-01-14:
Eine Implementierung in DAPNET, dass jeder Sender alle 10 Minuten sein Rufzeichen aussendet, erscheint mir zu kompliziert. Dies kann in die Eigenverantwortung des Senderbetreibers gelegt werden.
Ich schlage eine Textbox vor, in die ein Bakentext (z.B. "DB0WA FUNKRUF") eingetragen werden kann. Ein Timer prüft alle 10 Minuten, ob ein Bakentext vorhanden ist und sendet diesen an irgendeine RIC (z.B. 1234567). Daneben könnte noch ein Button sein, der die Bake sofort aussendet.

7h0ma5, 2017-01-15:
Darüber habe ich mir auch schon viele Gedanken gemacht. Es gibt Argumente dafür und dagegen:

**Für einen Einbau in RustPager:**
- Der Funkrufsender identifiziert sich auch, wenn die Verbindung zum Master unterbrochen ist.
- Weniger Code im DAPNET-Core.

**Gegen einen Einbau in RustPager:**
- Mehr Code im RustPager: Der DAPNET-Core besitzt bereits Code zum Generieren von Nachrichten für Skyper-Rubriken, das unterscheidet sich nochmal von normalen POCSAG-Nachrichten. Auch das zeitgesteuerte Aussenden von Nachrichten soll bald im DAPNET-Core implementiert werden.
- Das Rufzeichen des Transmitters muss an zwei Stellen festgelegt werden, einmal in der DAPNET-Datenbank und einmal in der RustPager-Konfiguration.
- Wenn die Verbindung zum Master unterbrochen ist, werden keine Nachrichten ausgesendet, es ist also auch keine Identifizierung nötig. Bei manuellen Tests muss der testende Funkamateur dann natürlich das Rufzeichen zur Identifikation in den Testnachrichten mitsenden.
- Es kann neben RustPager auch noch andere Programme geben, dort müsste das dann auch implementiert werden.

dh3wr, 2017-01-16:
Ich denke auch, im Core wäre das besser untergebracht.

dj7lc, 2017-01-16:
Gut, dann lassen wir das. War nur eine Idee :-)

closed 2017-01-16 by dj7lc