rwth-afu/UniPager #20: LED-Ansteuerung für Connected/Running/OnAir

dh3wr, 2017-01-29:
Es wäre schön, wenn man GPIO-Ports definieren könnte, mit denen man Frontplatten-LEDs ansteuern könnte. Eine freie Auswahl der GPIO-Ports oder auch "null" für "nicht vorhanden/gewünscht" wäre super. Einmal für den C9000, aber auch für die RasPagerDigi-Variante.

dh3wr, 2017-02-07:
Funktionswunschliste:
- RustPager läuft
- RustPager ist mit Core verbunden
- RustPager sendet (wenn möglich)

dk4pa, 2017-03-06:
Wäre es evtl. auch möglich ein Nextion Display einzubinden?

dh3wr, 2017-03-07:
@dk4pa Ich habe mit @7h0ma5 gesprochen. Wenn dann als externes Programm, welches sich als Websocket-Client mit dem Rustpager verbindet und dann die Weboberfläche emuliert. Ich bestelle mir mal ein solche Display und einen USB->3,3V TTL Wandler. Magst du die Software schreiben? Es gibt hier https://github.com/eidheim/Simple-WebSocket-Server einen Websocket Client in C++.

dk4pa, 2017-03-07:
@dh3wr Würde ich sofort machen, leider habe ich null Erfahrung was C++ oder ähnliches angeht.

dh3wr, 2017-03-09:
@7h0ma5 Macht es Sinn, die GPIO-LEDs auch mit einem externen Programm zu steuern, dass sich als Websocket-Client verbindet?

dh3wr, 2017-03-09:
Nextion Issue ausgelagert: https://github.com/rwth-afu/RustPagerNextion

7h0ma5, 2017-03-09:
@dh3wr Das würde auf jeden Fall Sinn machen. Vor allem kann dann sichergestellt werden, dass die LED, die anzeigt, ob der RustPager läuft, auf jeden Fall ausgeht, auch wenn der RustPager durch irgendeinen Grund gekillt wird und die LED nicht selbst ausschalten kann. Und ich muss mir keine Gedanken zur Implementierung machen 😉 Ein Problem ist nur, dass dann je nach gewähltem Transmitter zwei Programme gleichzeitig auf die GPIO-Pins zugreifen. Da müsste man testen, ob das ohne Probleme möglich ist.

dh3wr, 2017-03-09:
@7h0ma5 ok Thomas, ich schreibe mal ein C++ Programm dazu

dh3wr, 2017-03-09:
LED Issue ausgelagert, damit hier zu. https://github.com/rwth-afu/RustPagerLED

closed 2017-03-09 by dh3wr