rwth-afu/UniPager #22: Wenn RasPager als Typ ausgewählt ist, meldet sich trotzdem "SDRPager" an

dh3wr, 2017-01-29:
Typ ausgewählt, Save Config, Restart, sogar sudo systemctl restart rustpager, aber trotzdem wird [SDRPager... gesendet zum DAPNET-Core

dh3wr, 2017-01-31:
Grund liegt hier src/connection.rs
Zeile 32:
`self.writer.write(b"[SDRPager v1.2-SCP-#2345678]\r\n").unwrap();`

7h0ma5, 2017-01-31:
Ja, das ist noch hard-coded aus den Anfangszeiten. Den Typ des Pagers kann man einfach aus der Config lesen und dann mitsenden. Das Problem ist, dass die aktuelle Version vom DAPNET-Core noch keine dynamischen Pager-Typen zulässt. Dort sind nur "RasPager", "PR430", "SDRPager" und "DV4mini" zulässig (siehe [RaspagerProtocol.java](https://github.com/DecentralizedAmateurPagingNetwork/Core/blob/master/src/main/java/org/dapnet/core/transmission/RaspagerProtocol.java#L83)).

Soweit ich weiß, ist es geplant, dass bei der neuen Version, bei der der Client die Verbindung aufbaut, ein beliebiger String als Typ mitgeschickt werden kann. Ein String anstelle eines enums würde da auch Sinn machen, da der DAPNET-Core eigentlich keine Unterscheidungen zwischen den einzelnen Typen machen muss, oder doch?

Um das hier zu beheben müssten jedenfalls im DAPNET-Core noch die Transmitter-Typen "C9000" und "STM32Pager" hinzugefügt oder aber der dynamische Typ-String implementiert werden.

7h0ma5, 2017-02-07:
Im Branch client-auth für die vom Client initiierte Verbindung ist das jetzt korrekt Implementiert.

closed 2017-02-07 by 7h0ma5