rwth-afu/UniPager #25: Panick wenn man auf Audio Sender schaltet

dh3wr, 2017-02-06:
Ich starte sowohl den master branch und schalte im WebIF auf Audio um. Dann kommt:
root@sdrpager:/home/pi/RustPager/target/release# ./rustpager 
RustPager 0.3.3
Copyright (c) 2017 RWTH Amateurfunkgruppe

This program comes with ABSOLUTELY NO WARRANTY.
This is free software, and you are welcome to redistribute
and modify it under the conditions of the GNU GPL v3 or later.
<https://www.gnu.org/licenses/gpl-3.0.txt>

INFO - Listening for 44.225.164.3 on port 1337.
INFO - Initializing audio transmitter...
INFO - Detected Raspberry Pi 3 Model B
thread '<unnamed>' panicked at 'called `Option::unwrap()` on a `None` value', /buildslave/rust-buildbot/slave/nightly-dist-rustc-cross-host-linux/build/src/libcore/option.rs:323
note: Run with `RUST_BACKTRACE=1` for a backtrace.
INFO - Client denied: 44.225.164.2

dh3wr, 2017-02-07:
In Version 0.3.2 geht es. Z.B. mit ~/RustPager/target/release auf 44.225.164.52
Wenn du die kaputte Version testen willst: 44.225.164.236 --> sudo systemctl start rustpager

7h0ma5, 2017-02-07:
Das liegt daran, dass RustPager die WiringPi-Bezeichnung der Pins verwendet und deshalb Pin 23 nicht existiert. Ich habe das in der Konfiguration jetzt zu Pin 4 geändert. Mit einem Link zur Pin-Nummerierung, wie in #26 vorgeschlagen, und einer besseren Fehlermeldung sollte das keine Probleme mehr machen.

dh3wr, 2017-02-07:
Laut http://pi4j.com/images/j8header-3b-large.png gibt es beim Pi3 Rev. B Pin Nummer 25 und insb. Pin Nummer 25, auf den ich jetzt umsteigen will aus Kabellängengründen. Ist die SW für Pi3 nicht vorbereitet?

7h0ma5, 2017-02-07:
Die GPIO-Pins für den Raspberry Pi 3 sind in der Tat noch nicht drin, da ich den GPIO-Treiber selbst geschrieben habe und nicht wusste, dass der Raspberry Pi 3 mehr Pins hat. Habe dafür einen neuen Issue aufgemacht (#27).

closed 2017-02-07 by 7h0ma5

dh3wr, 2017-02-07:
OK, mit dem Pin 4 geht's nun

7h0ma5, 2017-02-07:
Im master und client-auth Branch sind jetzt die Raspberry Pi 3 GPIO-Pins implementiert.