rwth-afu/UniPager #29: Verbindungsabbruch zum Master nach Senden

dh3wr, 2017-02-08:
**Kommt mehrfach vor, immer nach Sendungen. Ich versuche mal, das Log vom Core noch hinzuzufügen.**

info Received Message { id: 0, mtype: AlphaNum, speed: Baud(1200), addr: 1234, func: AlphaNum, data: "asda shdjka hsdjkha klsdhjka hsdjkhajkhsdkla" }

info Transmitting...

info Transmission completed.

info Received Message { id: 173, mtype: Numeric, speed: Baud(1200), addr: 2504, func: Numeric, data: "173200 080217" }

info Connection closed.

info Waiting 12 seconds until TimeSlot(2)...

info Trying to connect to master...

info Connection established.

info Set TimeSlots { 2367ABEF }

info Transmitting...

info Transmission completed.

dh3wr, 2017-02-08:
Hier das Core-Log:
19:22:00.001 [DefaultQuartzScheduler_Worker-10] INFO  org.dapnet.core.transmission.TransmissionManager - Time sent to all connected transmitters.
19:22:00.005 [nioEventLoopGroup-3-2] ERROR org.dapnet.core.transmission.ServerHandler - Closing connection: Unexpected response received.
19:22:00.005 [nioEventLoopGroup-3-2] INFO  org.dapnet.core.transmission.ServerHandler - Connection closed.
19:22:00.005 [nioEventLoopGroup-3-2] WARN  org.dapnet.core.transmission.ServerHandler - Client has 1 pending acks.
19:22:00.006 [Incoming-1,DAPNET0.9.3.2,db0sda] INFO  org.dapnet.core.cluster.RpcListener - UpdateTransmitterStatus extern_gm1200 to ERROR: OK
19:22:01.013 [nioEventLoopGroup-3-3] INFO  org.dapnet.core.transmission.ServerHandler - Accepted new connection from /44.225.164.236:39032
19:22:01.064 [Incoming-1,DAPNET0.9.3.2,db0sda] INFO  org.dapnet.core.cluster.RpcListener - UpdateTransmitterStatus extern_gm1200 to ONLINE: OK

Taronyu, 2017-02-08:
Ich habe momentan die Vermutung, dass das an der Verarbeitung der Sequenznummern im Core liegt. War bei dem Test mehr als ein Client mit dem Core verbunden? Das ist ja beim Senden der Zeitnachricht abgeschmiert und die geht ja immer an alle verbundenen Clients raus. Der Einfachheit halber speichere ich die Sequenznummer in der Nachricht ab. Wenn eine Nachricht aber an mehrere Clients raus geht, überschreiben die sich eventuell die Sequenznummern und das ACK passt dann nicht mehr.

dh3wr, 2017-02-09:
Ja, es waren mehrere Sender verbunden. Sollte nicht jeder Sender-Thread im Master seine eigenen Seq.-Nummern zählen und verwalten?

Taronyu, 2017-02-09:
Die Sequenznummern werden auch pro Client getrennt verwaltet, das war schon vorher so. Das Problem war hier wahrscheinlich die Zuordnung zu den Nachrichten. Die Nachrichten werden mittels eines separaten Encoders in dieses Textformat umgewandelt, der Encoder wusste aber nichts vom jeweiligen Client und kannte daher auch nicht die korrekte Sequenznummer. Deswegen hatte ich die Sequenznummer in die Nachricht geschrieben. Da der ganze Kram aber asynchron läuft, wurde die Sequenznummer bei geteilten Nachrichten überschrieben. Ich habe das nun doch so umgebaut, dass der Encoder weiß, welcher Client die Nachricht empfangen soll und somit auch die korrekte Sequenznummer verwendet wird. Damit müsste sich das Problem eigentlich behoben haben, sofern es wirklich daran lag.

dh3wr, 2017-02-09:
Geht jetzt hervorragend! Danke

closed 2017-02-09 by dh3wr