rwth-afu/UniPager #31: Nach Änderung der IP des Core und Save Config passiert nichts

dh3wr, 2017-02-09:
Gestartet, IP geändert, "Save config".
Log-Ausgabe
````Connected to RustPager.
info Config updated. Initiating restart.
info Waiting for the connection to terminate... 
````
Mehr schieht nicht.

dh3wr, 2017-02-09:
Update: Ich war wohl zu ungeduldig. Nach ca. 20 Sekunden kommt
````
error Connection failed.
info Waiting for the scheduler to terminate...
info Scheduler stopped.
info Trying to connect to master...
info Initializing audio transmitter...
info Detected Raspberry Pi 3 Model B
info Scheduler started.
info Connection established.
info Set TimeSlots { 0123456789ABCDEF }
````

7h0ma5, 2017-02-09:
Das liegt daran, dass man in Rust (noch) kein Timeout für den Verbindungsaufbau definieren kann, es wird also das Standard-Timeout vom Betriebsystem verwendet. Wenn der Verbindungsaufbau sofort abgebrochen wird, weil der Host, zu dem man sich Verbinden will, sofort die Verbindung zu macht, ist das kein Problem, nur wenn der Host nicht online ist oder gar nicht antwortet, kann das etwas länger dauern. Erst wenn der Verbindungsaufbau gescheitert ist, empfängt der Transmitter-Thread die Nachricht, den Verbindungsaufbau abzubrechen und sich mit dem neuen Server zu verbinden.

7h0ma5, 2017-11-26:
Rust bietet mittlerweile die Möglichkeit ein Timeout für den Verbindungsaufbau zu setzen. In der neusten Version habe ich das auf 10 Sekunden gestellt, man muss also jetzt maximal nur noch 10 Sekunden warten, wenn der Verbindungsaufbau noch hängt.

closed 2017-11-26 by 7h0ma5