rwth-afu/UniPager #33: Welcome Message of client-auth nicht mit aktueller Core master-Version kompatibel

dh3wr, 2017-02-23:
Ich habe soeben den Commit https://github.com/DecentralizedAmateurPagingNetwork/Core/commit/a749238f1bc5e4727f5cb14551df1af66a26a408 installiert und dazu den neuesten client-auth des Rastpagers versucht. Es kommt:
````
01:46:31.979 [nioEventLoopGroup-3-1] INFO  org.dapnet.core.transmission.ServerHandler - Accepted new connection from /44.225.164.233:58126
01:46:31.980 [nioEventLoopGroup-3-1] ERROR org.dapnet.core.transmission.ServerHandler - Closing connection: Invalid welcome message format.
01:46:31.981 [nioEventLoopGroup-3-1] INFO  org.dapnet.core.transmission.ServerHandler - Connection closed.
01:46:31.981 [nioEventLoopGroup-3-1] WARN  org.dapnet.core.transmission.TransmitterManager - Client has no associated transmitter object.
````
Ich glaube, Philipp hatte am Format der Welcome-Message gedreht, was die Versions- und Typ-Übermittlung angeht.

dh3wr, 2017-02-23:
@Taronyu Noch was: Vielleicht kann man zum debuggen die invalide Welcome-Message mal ausgeben? Dann findet man das evtl. schneller, auch bei späteren Tests

dh3wr, 2017-02-23:
So ein Quatsch von mir: Ich sollte schon im auth-Client branch nachschauen. Da steht in der gleichen Datei ab Zeile 93
````  
pub fn run(&mut self) -> Result<()> {
let version = env!("CARGO_PKG_VERSION");
let id = format!("[RustPager-{} v{} {} {}]\r\n",
self.id, version, self.call, self.auth);
````

Das ist ja schon besser, passt aber wohl nicht zu https://github.com/DecentralizedAmateurPagingNetwork/Core/blob/experimental/src/main/java/org/dapnet/core/transmission/ServerHandler.java Zeile 34-35
````
private static final Pattern authPattern = Pattern
.compile("\\[([/\\p{Alnum}]+) v(\\d+\\.\\d+[-#\\p{Alnum}]*) (\\p{Alnum}+) (\\p{Alnum}+)\\]");
````

Was ist eigentlich mit der Umsetzung von https://github.com/DecentralizedAmateurPagingNetwork/Core/issues/52 ? Macht das Sinn für euch?

Taronyu, 2017-02-23:
Das war ein Bug im Pattern, das "-" bei "Rustpager-" war nicht im Pattern enthalten und somit schlug das Matching fehl. Ist im `master` des Cores gefixt.

dh3wr, 2017-02-23:
Gleiche Fehlermeldung weiterhin

dh3wr, 2017-02-23:
Fixed in https://github.com/DecentralizedAmateurPagingNetwork/Core/commit/990da3bc7f32d24e3685d12c15b780c3f0507486 with the help of Thomas. Das Issue hatte auch hier eigentlich nichts verloren 👍 )

closed 2017-02-23 by dh3wr