rwth-afu/UniPager #34: Neuer Namen für RustPager

dh3wr, 2017-02-24:
Nachdem der Name RustPager ja nun nicht optimal ist, würde ich hier gerne mögliche Alternativen diskutieren.
Da die Software ja alle Hardware-Varianten unterstützt, würde ich etwas wie
UniversalPager oder UnifiedPager
oder ähnlich vorschlagen.

7h0ma5, 2017-03-14:
Liste der bisherigen Vorschläge:

- UniversalPager
- UnifiedPager
- UniPager
- HamPocPager (von @MicManders auf Twitter)
- HAMpager (von @FaLanKeFu auf Twitter)
- AllPager (von @DK4PA auf Twitter)
- PocPager (von @DK4PA auf Twitter)

7h0ma5, 2017-03-20:
Der neue Name von RustPager wird **UniPager**. Alle nötigen Änderungen sind schon im debian-Branch vorhanden, mit dem gleichzeitig eine wesentlich einfachere Installation über ein APT-Repository möglich wird.

dh3wr, 2017-03-20:
Ok, ich ändere den CQDL Beitrag entsprechend

commit e730fbeb (2017-03-22): Rename RustPager to UniPager (close #34)
closed 2017-03-22 by 7h0ma5