rwth-afu/UniPager #36: Rustpager verliert ohne Fehlermeldung Verbindung zum Master nach IP-Adressewechsel

daccle, 2017-02-28:


dh3wr, 2017-02-28:
@daccle Ist das jetzt durch https://github.com/DecentralizedAmateurPagingNetwork/Core/issues/58 geklärt?

daccle, 2017-02-28:
Ich kann es erst heute Abend in Ruhe anschauen und schließe dann ggf. Den issue.

Am 28. Februar 2017 14:47:44 MEZ schrieb Ralf Wilke DH3WR <notifications@github.com>:
>@daccle Ist das jetzt durch
>https://github.com/DecentralizedAmateurPagingNetwork/Core/issues/58
>geklärt?

-- 
Diese Nachricht wurde von meinem Android-Mobiltelefon mit K-9 Mail gesendet.

dh3wr, 2017-03-05:
Ich habe gerade folgende Email bekommen. Sollte das mit diesem Commit auch gelöst sein?

Hei Ralf,

bei mir ist es etwas anders da kein VPN im Spiel ist, und daher die
Verbindung nicht über ein im RasPi endendes Tunnelinterface geht,
sondern direkt via Fritzbox, Telekoms RAS-Server zum Core bei euch.

Alle anderen wärend des Zwangstrennungsvorgangs laufenden Verbindungen
wie z.B. der streamripper, der einen Internetradiosender aufzeichnet
oder VLC, mit dem ich einen Stream höre brechen ab, retryen und machen
weiter, sobald die Verbindung wieder da ist.
Gibt es keine "keep alive"-Überwachung seitens des rustpagers, ob der
Core überhaput noch da ist bzw. die Verbindung zu ihm wieder da ist?

Viele Grüsse,
Jens

7h0ma5, 2017-03-05:
Ja, das müsste mit dem Commit gelöst sein. Damit wird die Keepalive-Extension von TCP aktiviert, das Betriebssystem ist dann dafür verantwortlich, die Keepalive-Pakete zu senden.

dh3wr, 2017-03-05:
Jens schreibt: Ein RustPager (db0prt) ist 0.4.0, dm0ai ist 0.3.3. Beide Stationen haben das Verbindungsproblem

7h0ma5, 2017-03-05:
Beim RustPager 0.3.3 ist das TCP-Keepalive noch deaktiviert. Bei der Versionsangabe `0.4.0` weiß man natürlich nicht, ob da der Commit https://github.com/rwth-afu/RustPager/commit/3482c0ccb85768f551d34c1f4915b213fd3e8f08 schon mit drin ist, weil es bisher noch kein neues Release gab. Ich habe das eben nochmal getestet und mit WireShark überprüft, das Keepalive funktioniert definitiv. 

Das Standard-Keepalive-Intervall von Linux ist auf 75 Sekunden gesetzt. So lange kann es im Worst-Case dauern, bis erkannt wird, dass der Master nicht mehr erreichbar ist. Mit `sudo sysctl net.ipv4.tcp_keepalive_intvl=10` kann man das auf 10 Sekunden setzen. Um das dauerhaft zu ändern, muss man das in der `/etc/sysctl.conf` eintragen.

daccle, 2017-03-06:
So, ich habe noch einmal ein git pull gemacht und kompiliere gerade. Dann schaue ich mal.

7h0ma5, 2017-03-14:
Das Problem scheint mit der neuen Version bisher nicht mehr aufgetreten zu sein.

commit 3482c0cc (2017-02-28): Enable TCP keepalive (should fix #36)
closed 2017-03-14 by 7h0ma5