rwth-afu/UniPager #41: /dev/ttyUSBx als PTT device für Audio Slaves

phl0, 2017-02-28:
Gerade habe ich DB0DDE auf RustPager wechseln wollen und stelle fest, dass die Audio Variante nur GPIOs als PTT unterstützt. Wird es auch (wieder) eine DTR/CTS Pin-Steuerung per /dev/ttyUSBx geben?
Das wäre cool, weil es mir zwei weitere Gänge auf den Turm ersparen würde ;-)

dh3wr, 2017-03-01:
Für DB0KWE brauche ich das auch. Hier ist ein x86er mit /dev/ttyS1 zu versorgen.

phl0, 2017-03-01:
Ich probiere mich mal an dem Rust Code ... Vielleicht klappt das ja mit ein wenig Try+Error :-)

phl0, 2017-03-01:
Ich hab mal (ganz langsam) angefangen:

https://github.com/phl0/RustPager/tree/devel

phl0, 2017-03-08:
Gerade festgestellt, dass wenn die GPIO Initialisierung fehlschlägt (weil z.B. x86 Plattform) auch der Audio output nicht geht.
Für PTT via Serial hab ich zumindest schon mal einen PoC.

phl0, 2017-03-08:
@dh3wr Kannst du mal meinen branch testen? Der sollte auch auf x86  laufen und die PTT-Steuerung per RS-232 erlauben. Beim Kompilieren gibt es noch eine Warnung. Daran arbeite ich noch. 
Der Code ist hier:

https://github.com/phl0/RustPager/tree/twoTransmitters

PR folgt, wenn alles sauber baut.

phl0, 2017-03-08:
Aktuellen Code-Stand aus meinem Branch läuft seit 10 Minuten bei DB0DDE im Testbetrieb. Bis auf die Warnung beim Kompilieren sieht es bis jetzt gut aus.

phl0, 2017-03-08:
Compiler Warnung ist auch gefunden und eliminiert :-)

dh3wr, 2017-03-09:
@phl0 Ich habe diese Woche leider weder Zeit noch Test Equipment. Sorry. Vielleich kann der Erschaffer @7h0ma5 ja etwas zu deinen Code-Ergänzungen sagen?

phl0, 2017-03-09:
Macht nix. Ich hab mittlerweile 2 Systeme mit dem Code laufen. Bisher problemlos. 
Ich warte mal auf Rückmeldung vom Autor. Andernfalls mache ich einfach n PR.

phl0, 2017-03-09:
Hab mal einen PR gemacht: https://github.com/rwth-afu/RustPager/pull/49

commit 217907aa (2017-03-14): Implement GPIO/Serial PTT configuration (close #41)
closed 2017-03-14 by 7h0ma5