rwth-afu/UniPager #42: Compilieren auf Banana Pi M1 mit Raspbina Lite (Jessy) geht nicht

dg1kwa, 2017-03-05:
Ich kann den Rustpager auf einen Banana Pi M1 nicht compilieren. Image ist ganz frisch:
Raspbian 	Lite 	3.4 	2016-07-12 von der Homepage installiert.

Ich habe nur GIT nachinstalliert, das passiert dann beim compilieren

       ......
       Fresh num-integer v0.1.32
       Fresh slab v0.1.3
       Fresh num-iter v0.1.32
       Fresh raspi v0.1.0 (file:///usr/local/src/RustPager/lib/raspi)
       Fresh num v0.1.36
       Fresh log v0.3.6
       Fresh chrono v0.2.25
       Fresh mio v0.5.1
       Fresh ws v0.5.3
       Fresh tiny_http v0.5.7
   Compiling rustpager v0.4.0 (file:///usr/local/src/RustPager)
     Running `rustc --crate-name rustpager src/main.rs --crate-type bin --emit=dep-info,link -C opt-level=3 -C metadata=8dd5da31e9f                    3e1fb -C extra-filename=-8dd5da31e9f3e1fb --out-dir /usr/local/src/RustPager/target/release/deps -L dependency=/usr/local/src/RustP                    ager/target/release/deps --extern lazy_static=/usr/local/src/RustPager/target/release/deps/liblazy_static-39d04338418c9a4e.rlib --e                    xtern serde=/usr/local/src/RustPager/target/release/deps/libserde-b116c413b6ec9a4b.rlib --extern serde_derive=/usr/local/src/RustPa                    ger/target/release/deps/libserde_derive-0b1a0eaf28eed447.so --extern serial=/usr/local/src/RustPager/target/release/deps/libserial-                    2ae09162fcdfddf8.rlib --extern ws=/usr/local/src/RustPager/target/release/deps/libws-1c06afb72f34ff59.rlib --extern raspi=/usr/loca                    l/src/RustPager/target/release/deps/libraspi-c33bc4cdffe26395.rlib --extern net2=/usr/local/src/RustPager/target/release/deps/libne                    t2-8d6db02178fc0f61.rlib --extern serde_json=/usr/local/src/RustPager/target/release/deps/libserde_json-a0b08984b58def3f.rlib --ext                    ern log=/usr/local/src/RustPager/target/release/deps/liblog-973d2302a68c7b2e.rlib --extern tiny_http=/usr/local/src/RustPager/targe                    t/release/deps/libtiny_http-90c763dea07c1386.rlib`
error: proc-macro derive panicked
  --> src/status.rs:10:10
   |
10 | #[derive(Serialize, Clone, Copy, Debug)]
   |          ^^^^^^^^^
   |
   = help: message: already mutably borrowed: BorrowError

error: Could not compile `rustpager`.

Caused by:
  process didn't exit successfully: `rustc --crate-name rustpager src/main.rs --crate-type bin --emit=dep-info,link -C opt-level=3                     -C metadata=8dd5da31e9f3e1fb -C extra-filename=-8dd5da31e9f3e1fb --out-dir /usr/local/src/RustPager/target/release/deps -L dependen                    cy=/usr/local/src/RustPager/target/release/deps --extern lazy_static=/usr/local/src/RustPager/target/release/deps/liblazy_static-39                    d04338418c9a4e.rlib --extern serde=/usr/local/src/RustPager/target/release/deps/libserde-b116c413b6ec9a4b.rlib --extern serde_deriv                    e=/usr/local/src/RustPager/target/release/deps/libserde_derive-0b1a0eaf28eed447.so --extern serial=/usr/local/src/RustPager/target/                    release/deps/libserial-2ae09162fcdfddf8.rlib --extern ws=/usr/local/src/RustPager/target/release/deps/libws-1c06afb72f34ff59.rlib -                    -extern raspi=/usr/local/src/RustPager/target/release/deps/libraspi-c33bc4cdffe26395.rlib --extern net2=/usr/local/src/RustPager/ta                    rget/release/deps/libnet2-8d6db02178fc0f61.rlib --extern serde_json=/usr/local/src/RustPager/target/release/deps/libserde_json-a0b0                    8984b58def3f.rlib --extern log=/usr/local/src/RustPager/target/release/deps/liblog-973d2302a68c7b2e.rlib --extern tiny_http=/usr/lo                    cal/src/RustPager/target/release/deps/libtiny_http-90c763dea07c1386.rlib` (exit code: 101)
root@bpi-rustpager:/usr/local/src/RustPager#

7h0ma5, 2017-03-05:
Da ich keinen Banana Pi besitze, kann ich das Problem leider nicht reproduzieren. Das Problem liegt aber beim Compiler, eventuell hilft ein Update des Compilers mit `rustup update`.

Eine andere Möglichkeit wäre, die fertig kompilierte Version von RustPager zu verwenden. Für den Banana Pi M1 müsste das die `armv7hf`-Version sein, welche man hier herunterladen kann: https://github.com/rwth-afu/RustPager/releases/tag/v0.4.0

dg1kwa, 2017-03-06:
> thread [2].
> Links:

also das Image ist komplett frisch, danach apt-get update und upgrade 
gemacht.

Dann habe ich alles frisch installiert.

Aber ich teste mal eine fertige Version.

73 de Andreas

dg1kwa, 2017-03-07:
also das rustup hilft nicht, aber das fertige Paket läuft.

Ist mir sogar fast lieber weil es auch deutlich schneller geht :)

phl0, 2017-03-09:
Kompilieren unter Jessie Lite auf einem Raspberry Pi 2 / 3 / B+ klappt ...

dh3wr, 2017-03-13:
Auf einem Pi3 B geht es auch nicht.
````
pi@c9000:~/RustPager $ cargo build --release
   Compiling rustpager v0.4.0 (file:///home/pi/RustPager)
error: proc-macro derive panicked
  --> src/status.rs:10:10
   |
10 | #[derive(Serialize, Clone, Copy, Debug)]
   |          ^^^^^^^^^
   |
   = help: message: already mutably borrowed: BorrowError

error: Could not compile `rustpager`.

To learn more, run the command again with --verbose.
````
OS ist ein frisches jessie lite. Vielleicht hilft das: http://stackoverflow.com/questions/40196586/mutable-reference-to-container-object-within-iterator-loop ?

dh3wr, 2017-03-13:
Wenn du testen willst: 44.225.164.229 pi password wie immer

dg1kwa, 2017-03-13:
gleiche Meldung wie auf meiner Banane

dh3wr, 2017-03-13:
Kompiler-Sache. Erst alter Kompiler: Alles gut. Dann ''rustup update''. Schon gehts nicht mehr

''''
   Compiling rustpager v0.4.0 (file:///home/pi/RustPager)
    Finished release [optimized] target(s) in 1107.91 secs

**pi@raspagerdf6ef:~/RustPager $ rustup update**

info: syncing channel updates for 'nightly-armv7-unknown-linux-gnueabihf'
282.9 KiB / 282.9 KiB (100 %) 199.4 KiB/s ETA:   0 s
info: downloading component 'rustc'
 41.0 MiB /  41.0 MiB (100 %) 780.8 KiB/s ETA:   0 s
info: downloading component 'rust-std'
 54.9 MiB /  54.9 MiB (100 %) 688.0 KiB/s ETA:   0 s
info: downloading component 'cargo'
  8.2 MiB /   8.2 MiB (100 %) 764.8 KiB/s ETA:   0 s
info: installing component 'rustc'
info: installing component 'rust-std'
info: installing component 'cargo'
info: checking for self-updates

  nightly-armv7-unknown-linux-gnueabihf updated - rustc 1.17.0-nightly (824c9ebbd 2017-03-12)

pi@raspagerdf6ef:~/RustPager $ cargo build --release
   Compiling matches v0.1.4

[...]

   Compiling ws v0.6.0
   Compiling rustpager v0.4.0 (file:///home/pi/RustPager)
error: proc-macro derive panicked
  --> src/status.rs:10:10
   |
10 | #[derive(Serialize, Clone, Copy, Debug)]
   |          ^^^^^^^^^
   |
   = help: message: already mutably borrowed: BorrowError

error: Could not compile `rustpager`.

To learn more, run the command again with --verbose.
pi@raspagerdf6ef:~/RustPager $
''''

phl0, 2017-03-14:
Ich bekomme gerade den selben Fehler auf einem Raspberry Pi 2B mit Debian Jessie. Liegt es an der Version von rustc? Wie finde ich dessen Version heraus bzw. wie kann man den ggf. downgraden?

phl0, 2017-03-14:
Ich bin tageweise zurück bis zur Version 

nightly-2017-02-25-armv7-unknown-linux-gnueabihf installed - rustc 1.17.0-nightly (08230775a 2017-02-24)

Bis jetzt immer noch der selbe Fehler. Probiere weiter

phl0, 2017-03-15:
df2et@pagerpi:~/RustPager $ rustup toolchain install nightly-2017-01-25
info: syncing channel updates for 'nightly-2017-01-25-armv7-unknown-linux-gnueabihf'
info: downloading component 'rustc'
 46.4 MiB /  46.4 MiB (100 %)   6.0 MiB/s ETA:   0 s                
info: downloading component 'rust-std'
 61.4 MiB /  61.4 MiB (100 %)   4.1 MiB/s ETA:   0 s                
info: downloading component 'cargo'
  4.3 MiB /   4.3 MiB (100 %) 1010.9 KiB/s ETA:   0 s                
info: installing component 'rustc'
info: installing component 'rust-std'
info: installing component 'cargo'

  nightly-2017-01-25-armv7-unknown-linux-gnueabihf installed - rustc 1.16.0-nightly (83c2d9523 2017-01-24)

df2et@pagerpi:~/RustPager $ rustup toolchain install nightly-2017-01-29
info: syncing channel updates for 'nightly-2017-01-29-armv7-unknown-linux-gnueabihf'
Total: 633 B Speed:   0 B/s                error: no release found for 'nightly-2017-01-29'
df2et@pagerpi:~/RustPager $ cargo build --release                                                                                           
   Compiling rustpager v0.4.0 (file:///home/df2et/RustPager)
error: proc-macro derive panicked
  --> src/status.rs:10:10
   |
10 | #[derive(Serialize, Clone, Copy, Debug)]
   |          ^^^^^^^^^
   |
   = help: message: already mutably borrowed: BorrowError

error: Could not compile `rustpager`.

To learn more, run the command again with --verbose.

phl0, 2017-03-15:
```
df2et@pagerpi:~/RustPager $ rustup toolchain install nightly-2016-12-02
info: syncing channel updates for 'nightly-2016-12-02-armv7-unknown-linux-gnueabihf'
142.4 KiB / 142.4 KiB (100 %)  69.2 KiB/s ETA:   0 s                
info: downloading component 'rustc'
 51.6 MiB /  51.6 MiB (100 %)   6.6 MiB/s ETA:   0 s                
info: downloading component 'rust-std'
 65.0 MiB /  65.0 MiB (100 %)   3.7 MiB/s ETA:   0 s                
info: downloading component 'cargo'
  4.2 MiB /   4.2 MiB (100 %)   1.0 MiB/s ETA:   0 s                
info: installing component 'rustc'
info: installing component 'rust-std'
info: installing component 'cargo'

  nightly-2016-12-02-armv7-unknown-linux-gnueabihf installed - rustc 1.15.0-nightly (908dba0c9 2016-12-01)

df2et@pagerpi:~/RustPager $ cargo build --release
   Compiling rustpager v0.4.0 (file:///home/df2et/RustPager)
error: proc-macro derive panicked
  --> src/status.rs:10:10
   |
10 | #[derive(Serialize, Clone, Copy, Debug)]
   |          ^^^^^^^^^
   |
   = help: message: already mutably borrowed: BorrowError

error: Could not compile `rustpager`.

To learn more, run the command again with --verbose.
```

dg1kwa, 2017-03-16:
Ich habe das gestern auf meinem Raspi 1 bei DB0KOE auch beobachtet.

Diesesmal konnte ich ohne Probleme compilieren, allerdings liefen auch 
einige Rust-Updates durch.

73 de Andreas

Raspbian Lite (Jessy) geht nicht (#42)

dh3wr, 2017-03-16:
nightly-armv7-unknown-linux-gnueabihf updated - rustc 1.17.0-nightly (0aeb9c129 2017-03-15)
Ebenfalls der gleiche Fehler.

dg1kwa, 2017-03-16:
komisch auf meinem Pi1 geht es ohne Probleme ..

phl0, 2017-03-18:
Kann es sein, dass es nur bei ARMv7 Compilern auftritt?
Bei mir klappt es auf einem RPi 1B und einem RPi B+. Das sind beides ARMv6 CPUs....

MurrayBozinsky, 2017-03-18:
Der vorkompilierte RustPager 0.42 läuft auf meinem RPi3B einwandfrei. 
In welches Verzeichnis kopiere ich die lauffähige Datei, um sie als Dienst
automatisch starten lassen kann und wie bewerkstelle ich das ?

phl0, 2017-03-18:
Siehe https://github.com/rwth-afu/RustPager/blob/master/README.md
Das Binary gehört nach /usr/local/bin

dg1kwa, 2017-03-19:
Kann ich bestätigen. Auf dem Arm 6 OK auf dem Arm 7 klappt es nicht. 

⁣Gesendet mit TypeApp ​

phl0, 2017-03-28:
Gerade ein `rustup update` und `cargo update` gemacht. Jetzt kompilierte es auch auch einem AVMv7 ... ARMv6 ging auch.

dh3wr, 2017-04-24:
Da es ja jetzt die repository-Variante gibt, können wir das zumachen, oder?

closed 2017-04-24 by 7h0ma5