rwth-afu/UniPager #44: Minor: Frequenzkorrektur hat als Default-Wert den Modulation-Registerwert 13 statt -67

dh3wr, 2017-03-05:
Sascha schreibt:

Die Standard-Vorgabe bei der Frequenzkorrektur im Rustpager ist "13". Ich weiss nicht, ob das Absicht oder ein Versehen ist.

Der funktionierende Wert, wie er auch im Raspager-Handbuch von DF6EF festgehalten ist, ist ja "-67", "13" wiederum ist der Standardwert für die FSK-Bandbreite. Wenn man hier nicht aufpasst, sucht man erst mal kurz, warum es nicht funktioniert.

7h0ma5, 2017-03-05:
Die Frequenzkorrektur ist standardmäßig auf 0 eingestellt (siehe https://github.com/rwth-afu/RustPager/blob/master/src/config.rs#L32). Der Wert muss also vom Benutzer schon mal geändert worden sein. Da sich der Korrekturwert bei jedem Raspager unterscheidet (bei dem, mit dem ich teste, sind es zum Beispiel -43), macht es keinen Sinn, irgendetwas vorzugeben.

closed 2017-03-05 by 7h0ma5

dh3wr, 2017-03-05:
Ich informiere den betroffenen OM.