rwth-afu/UniPager #52: Rustpager Audio liefert panick

dh3wr, 2017-03-13:
Von DL8SDL erhalten:
````
Hei Ralf,

eben begrüsste mich folgende Fehlermeldung (Sender db0prt,
RustPagerAudio):

INFO - Received Message { id: 81, mtype: Numeric, speed: Baud(1200),
addr: 2504, func: Numeric, data: "141800   100317" }
Playing raw data 'stdin' : Unsigned 8 bit, Rate 48000 Hz, Mono
WARN - Disconnected from master.
INFO - Trying to connect to master...
ERROR - Connection failed.
INFO - Trying to connect to master...
ERROR - Connection failed.
INFO - Trying to connect to master...
thread '<unnamed>' panicked at 'called `Result::unwrap()` on an `Err`
value: Error { repr: Os { code: 107, message: "Transport endpoint is not
connected" } }', /checkout/src/libcore/result.rs:860
note: Run with `RUST_BACKTRACE=1` for a backtrace.

Viele Grüsse,
Jens
````

7h0ma5, 2017-03-13:
Ohne Backtrace kann ich leider nicht sagen, wo genau das Problem auftritt. Es ist auf jeden Fall ein Fehler vom Betriebsystem, bei dem keine vernünftige Fehlerbehandlung möglich ist. Ich vermute es liegt ein Problem mit einem gemounteten Dateisystem vor, das nicht mehr verfügbar ist. Diese Fehlermeldung wird normalerweise nur von einem FUSE-Dateisystem wie sshfs oder ntfs-3g erzeugt. Es könnte auch ein Problem mit der SD-Karte vorliegen.

dk4pa, 2017-03-13:
Hallo Zusammen,

gleiches konnte ich eben auch bei mir feststellen.
INFO - Received Message { id: 151, mtype: Numeric, speed: Baud(1200), addr: 2504, func: Numeric, data: "104100   130317" }
Wiedergabe: Rohdaten 'stdin' : Unsigned 8 bit, Rate: 48000 Hz, mono
thread '<unnamed>' panicked at 'called `Result::unwrap()` on an `Err` value: Error { repr: Os { code: 107, message: "Transport endpoint is not connected" } }', /checkout/src/libcore/result.rs:860
note: Run with `RUST_BACKTRACE=1` for a backtrace.

Gruß
Stefan, DK4PA

dh3wr, 2017-03-15:
Jens fragt, wie man denn den Backtrace anmacht. Any help?

phl0, 2017-03-15:
Run with RUST_BACKTRACE=1 for a backtrace würde ich sagen?!

dh3wr, 2017-03-15:
Als Kommandozeilenparameter oder Umgebungsvariable?

7h0ma5, 2017-03-15:
Hier hatte ich schonmal was dazu geschrieben: https://github.com/rwth-afu/RustPager/issues/9#issuecomment-271659616

dh3wr, 2017-03-15:
Habs weitergegeben. Danke!

MurrayBozinsky, 2017-03-18:
pi@DM0AI-7:~ $ sudo systemctl enable rustpager
Failed to execute operation: No such file or directory

Fehlt da rustpager.service ? Was muss da drin stehen ?
Bei dem vorkompilierten RustPager habe ich ja kein install.sh

phl0, 2017-03-18:
Dann nimm doch das install.sh aus dem github repo bzw. tu das, was da drin steht.

7h0ma5, 2017-03-24:
Das Problem sollte mit Commit cadaf63114c9ad271a4fa4672fa184ededa61832 behoben sein.

closed 2017-03-24 by 7h0ma5