rwth-afu/UniPager #53: Release 0.4.1 läuft nicht auf "ARMv6-compatible processor rev 7 (v6l)"

dh3wr, 2017-03-13:
Egal welche *arm* Variante ich ausprobiere, es kommt genau nichts.
````
pi@raspagerc9000:~ $ tar fxvz rustpager-armv7hf.tar.gz
rustpager
pi@raspagerc9000:~ $ sudo ./rustpager
pi@raspagerc9000:~ $
````
````
pi@raspagerc9000:~ $ ps ax | grep rustpager
28807 pts/1    S+     0:00 grep --color=auto rustpager
````

````
 cat /proc/cpuinfo
processor       : 0
model name      : ARMv6-compatible processor rev 7 (v6l)
BogoMIPS        : 697.95
Features        : half thumb fastmult vfp edsp java tls
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xb76
CPU revision    : 7

Hardware        : BCM2708
Revision        : 000d
Serial          : 00000000c932870c
````

7h0ma5, 2017-03-13:
Der Raspberry Pi 1 braucht eine Spezialversion, weil der nur Thumb-1 kann. Blöderweise kann man den ARM-GCC-Crosscompiler unter Ubuntu nicht dazu zwingen, ARM-Code und keinen Thumb-2-Code auszuspucken. Man muss den Cross-Compiler selbst kompilieren, damit das funktioniert. Ich lade sobald wie möglich die Spezialversion für den Raspberry Pi 1 hoch.

closed 2017-03-13 by 7h0ma5

7h0ma5, 2017-03-14:
rustpager-rpi1.tar.gz ist jetzt verfügbar für Version 0.4.1