rwth-afu/UniPager #55: Zeitschlitze hintereinander

dk4pa, 2017-03-15:
Hallo, 
ich habe bei db0iga die Zeitschlitze AB aktiv gehabt. Hier ist es allerdings so, dass scheinbar immer nur der erste Zeitschlitz A aktiv wird. B kommt nicht zu tragen. Es müsste doch, wenn Zeitschlitz A nicht mehr aktiv, aber ein Ruf abgesetzt wird der B zu tragen kommen?

So bekomme ich immer nur "Waiting xx seconds until TimeSlot(A)...

Nun habe ich mal die Zeitschlitze angepasst db0iga (AE), dk4pa (BF).
Hier kommt nun auch der zweite Zeitschlitz zum Einsatz. 
Ist das so normal, setzt man den Abstand der Zeitschlitze besser etwas auseinander?

vy73, DK4PA Stefan

7h0ma5, 2017-03-15:
Hallo Stefan,

ich habe es gerade nochmal getestet, es wird definitiv auch gesendet, wenn die Nachricht reinkommt und gerade ein Zeitschlitz in Folge eines anderen aktiv ist. Die Nachricht "Waiting xx seconds until" kommt dann natürlich nicht mehr, weil das Warten ja nicht notwendig ist.

Die Log-Ausgabe "Transmitting..." gibt es jetzt nicht mehr, da dafür dann unter Status "ON AIR" rot wird. Je nach ausgewähltem Transmitter kann "ON AIR" auch nur sehr kurz angehen, daran wird noch gearbeitet.

Am besten du testest nochmal mit einem Pager oder noch besser einem RTL-SDR-Stick, ob die Rufe wirklich rausgehen.

73, Thomas DL2IC

dk4pa, 2017-03-15:
Hallo Thomas,

kann ich bestätigen, danke für den Tipp. Ich werde allerdings den Abstand der Zeitschlitze belassen, so dauert es nicht so lange bis zur nächsten Aussendung.
Kann somit geschlossen werden, sorry für die Mühe.

closed 2017-03-15 by dk4pa