rwth-afu/UniPager #56: WebSocket mit curl

dh3wr, 2017-03-16:
Moin,
hat jemand eine Idee, warum ``curl -i -N -H "Connection: Upgrade" -H "Upgrade: websocket" -H "Host: c9000.db0sda.ampr.org:8055" -H "Origin: http://db0sda.ampr.org" -d '{"GetVersion" } ' http://c9000.db0sda.ampr.org:8055`` 400 liefert?

dh3wr, 2017-03-16:
````
HTTP/1.1 400 Bad Request

WebSocket Protocol Error: Unable to parse WebSocket key.root
````

7h0ma5, 2017-03-16:
Probier mal `curl -i -N -H "Connection: Upgrade" -H "Upgrade: websocket" -H "Sec-Websocket-Key: test" -H "Sec-Websocket-Version: 13" -d '"GetVersion"' http://c9000.db0sda.ampr.org:8055`, aber ich glaube man kann mit CURL keine Websocket-Frames senden.

dh3wr, 2017-03-17:
Danke, geht so halb. Das Kommando wird zwar nicht ausgeführt, aber ich bekommen die Server-Meldungen..
````
curl -i -N -H "Connection: Upgrade" -H "Upgrade: websocket" -H "Sec-Websocket-Key: test" -H "Sec-Websocket-Version: 13" -d '"GetVersion"' http://c9000.db0sda.ampr.org:8055
HTTP/1.1 101 Switching Protocols
Connection: Upgrade
Sec-WebSocket-Accept: tNpbgC8ZQDOcSkHAWopKzQjJ1hI=
Upgrade: websocket

▒~▒{"Status":{"connected":true,"transmitting":true,"timeslots":[false,false,false,false,true,true,false,false,false,false,false,false,true,true,false,false]}}▒~▒{"Status":{"connected":true,"transmitting":false,"timeslots":[false,false,false,false,true,true,false,false,false,false,false,false,true,true,false,false]}}▒~▒{"Log":[3,"Received Message { id: 157, mtype: Numeric, speed: Baud(1200), addr: 2504, func: Numeric, data: \"095900   170317\" }"]}▒5{"Log":[3,"Waiting 18 seconds until TimeSlot(4)..."]}▒~▒{"Status":{"connected":true,"transmitting":true,"timeslots":[false,false,false,false,true,true,false,false,false,false,false,false,true,true,false,false]}}▒~▒{"Status":{"connected":true,"transmitting":false,"timeslots":[false,false,false,false,true,true,false,false,false,false,false,false,true,true,false,false]}}
````

dh3wr, 2017-03-17:
Also die Kommandos werden laut tcpdump nicht im Klartext übertragen. Ich machen dann einfach für Nagios eine kleines C++ Programm. https://github.com/rwth-afu/RustPagerNagios

closed 2017-03-17 by dh3wr