rwth-afu/UniPager #60: Install mit apt-get liefert --> Unescaped left brace in regex is deprecated <--

dh3wr, 2017-03-21:
````
pi@c9000:/etc/apt/sources.list.d $ sudo apt-get install unipager
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following NEW packages will be installed:
  unipager
0 upgraded, 1 newly installed, 0 to remove and 344 not upgraded.
Need to get 614 kB of archives.
After this operation, 1,703 kB of additional disk space will be used.
Get:1 http://db0sda.ampr.org/debian unipager/main armhf unipager armhf 0.4.2+0~20170320005227.38~1.gbp2d475f [614 kB]
Fetched 614 kB in 6s (96.1 kB/s)
Unescaped left brace in regex is deprecated, passed through in regex; marked by <-- HERE in m/^(.*?)(\\)?\${ <-- HERE ([^{}]+)}(.*)$/ at /usr/share/perl5/Debconf/Question.pm line 72.
Unescaped left brace in regex is deprecated, passed through in regex; marked by <-- HERE in m/\${ <-- HERE ([^}]+)}/ at /usr/share/perl5/Debconf/Config.pm line 30.
Selecting previously unselected package unipager.
(Reading database ... 51466 files and directories currently installed.)
Preparing to unpack .../unipager_0.4.2+0~20170320005227.38~1.gbp2d475f_armhf.deb ...
Unpacking unipager (0.4.2+0~20170320005227.38~1.gbp2d475f) ...
Setting up unipager (0.4.2+0~20170320005227.38~1.gbp2d475f) ...
Created symlink from /etc/systemd/system/multi-user.target.wants/unipager.service to /lib/systemd/system/unipager.service.
Unescaped left brace in regex is deprecated, passed through in regex; marked by <-- HERE in m/^(.*?)(\\)?\${ <-- HERE ([^{}]+)}(.*)$/ at /usr/share/perl5/Debconf/Question.pm line 72.
Unescaped left brace in regex is deprecated, passed through in regex; marked by <-- HERE in m/\${ <-- HERE ([^}]+)}/ at /usr/share/perl5/Debconf/Config.pm line 30.
pi@c9000:/etc/apt/sources.list.d $ 
````

7h0ma5, 2017-03-21:
Ist das auf dem Raspberry PI 3 aufgetreten, den ich auch zum Kompilieren benutzt habe? 

Damit das funktionierte musste ich ein paar Pakete aus dem Debian-Stretch Repository installieren, weil die nur da vorhanden waren, und deshalb wurde dort auch Perl auf eine neuere Version geupdated.

Dieses neue Perl beschwert sich dann bei der Installation des Pakets, weil das Paket mit den Skripten für Debian-Jessie erstellt wurde. Auf Debian-Jessie gibt es diese Fehlermeldungen deshalb nicht.

Solange Debian-Stretch noch in der Beta-Phase ist, kann man damit leben, aber sobald das draußen sollten die Pakete wohl standardmäßig für Stretch erstellt werden.

dh3wr, 2017-03-21:
Es ist auf 44.225.164.229 aufgetreten.

7h0ma5, 2017-03-21:
Ok, dann ist es genau so, wie ich beschrieben habe. Wenn wir das Kompilieren auf das neue System auslagern, können wir die Pakete auf diesem PI auch wieder downgraden.

closed 2017-03-21 by 7h0ma5