rwth-afu/UniPager #61: UniPager verliert im Dauerbetrieb die Verbindung

dj7lc, 2017-03-22:
Nach längerer Betriebszeit verliert der UniPager die Verbindung zu Dapnet und wird in der Übersicht mit "Error" gelistet. Ein Klick auf "Restart" im Webinterface behebt das Problem. Die Lösung könnte ein Cronjob oder Timer sein, der die "Restart"-Routine jede Nacht einmal ausführt.

Taronyu, 2017-03-23:
Da würde ich ehrlich gesagt erst einmal etwas Energie in die Diagnose stecken. Kann auch gut sein, dass der Fehler vom Core ausgeht.

MurrayBozinsky, 2017-03-23:
Hier (DM0AI und DB0PRT je am VDSL-Anschluss der .T...) verlieren die
(bislang noch Rust)Pager die Verbindung immer zur nächtlichen Zwangstrennung.
Ein morgendlicher Restart heilt alles, aber von 3:20 nachts bis zu $Aufstehzeit läuft kein Pagersender.

Taronyu, 2017-03-23:
Okay, Zwangstrennung hatte ich vergessen, das muss dann tatsächlich der
Client handhaben.

7h0ma5, 2017-03-24:
Mit der neusten Version im Debian-Repository sollte nach spätestens 2 Minuten die Verbindung wieder neu hergestellt werden. Bitte mit `sudo apt-get update` und `sudo apt-get upgrade` aktualisieren.

MurrayBozinsky, 2017-03-25:
Funktioniert bislang, die erste Zwangstrennung heute nacht wurde ohne ERROR überstanden

dh3wr, 2017-03-26:
Dann kann das zugemacht werden?

closed 2017-04-24 by 7h0ma5