rwth-afu/UniPager #65: Öffnen des Webinterface auf verschiedenen Rechnern erzeugt Versatz

dj7lc, 2017-03-24:
Wenn ich auf einem Rechner die Webinterfaces verschiedener Sender öffne, laufen die Zeitschlitze in allen Tabs und auch verschiedenen Browsern synchron. Auf jedem bisher getesteten Rechner trifft das zu. Stehen aber zwei verschiedene Rechner nebeneinander, fällt eine zeitliche Diskrepanz zwischen den Slot-Anzeigen auf.
Wenn das Webinterface nur auf einem Rechner geöffnet ist, wird genau dann gesendet, wenn die Zeitschlitzanzeige es vorgibt. Ist das Webinterface auf zwei verschiedenen Rechnern geöffnet, wird in der Mitte zwischen den verschiedenen Anzeigen gesendet. Eine Stoppuhr-Messung hat gezeigt, dass der Sender tatsächlich aus dem Takt kommt. Die On-Air-Anzeige stimmt immer.
Sobald einer der beiden Rechner den Browsertab schließt, ist der Spuk vorbei.
Dieser überaus komplexe Sachverhalt lässt sich in einem Wort treffend beschreiben: Häää?

7h0ma5, 2017-03-24:
Das liegt wohl daran, dass die Berechnung des aktuellen Zeitslots im Browser abläuft und daher von der aktuellen Systemzeit des Rechners, auf dem der Browser läuft, abhängt. Wenn zwei verschiedene Rechner unterschiedliche Systemzeiten haben, wird dann auf jedem Rechner auch ein unterschiedlicher aktiver Zeitslot angezeigt. Auswirkungen auf die interne Zeitslotberechnung von UniPager kann das aber eigentlich nicht haben. Es ist geplant, dass die Browser in Zukunft den aktuellen Zeitslot über den WebSocket bekommen und ihn nicht mehr selbst berechnen, damit es da keine Unterschiede mehr gibt.

dj7lc, 2017-03-24:
Das könnte die Lösung sein. Warten wir ab, ob dann auch die gegenseitige Beeinflussung wegfällt. Bis dahin können wir wieder zu machen.

closed 2017-03-24 by dj7lc

7h0ma5, 2017-03-24:
Die serverseitige Timeslot-Berechnung ist jetzt in der aktuellen Version im Debian-Repository implementiert, bitte damit noch mal testen.