rwth-afu/UniPager #66: Sendung geht über Timeslots hinaus oder Anzeige falsch

dh3wr, 2017-03-27:
Das sollte man doch eigentlich nicht sehen, oder? Version 0.5.1
[image]

dh3wr, 2017-03-27:
Geht sogar noch schlimmer
[image]

7h0ma5, 2017-03-27:
Das kann daran liegen, dass die TimeSlot-Implementierung noch nicht perfekt ist. Momentan wird nur getestet, ob der Start einer Nachricht noch im TimeSlot drin liegt. Die Berechnung der Übertragungsdauer wird aber komplett neu gemacht. So weit wie im 2. Bild sollte das aber eigentlich nicht gehen, höchstens 100 ms drüber.

dh3wr, 2017-03-27:
Tja, ist wohl aber so... http://44.225.164.234:8073/ :-)

7h0ma5, 2017-03-27:
Interessant, was für ein Skript benutzt du da zum Testen?

dh3wr, 2017-03-27:
````
#!/usr/bin/python3
import json
import pprint
import websocket
import time
from websocket import create_connection

websocket.enableTrace(True)
ws = create_connection('ws://44.225.164.234:8055/')

n = 1

while n <= 100:
    string_to_send = "{\"SendMessage\": {\"addr\": %s, \"data\": \"%s\"}}" % (n,n)
    print(string_to_send)
    ws.send(string_to_send)
    n = n + 1
    time.sleep( 1 )
````

dh3wr, 2017-06-20:
Ich mache hier nochmal eine Erinnerung dran, denn das Verhalten ist natürlich nicht so der Knaller.

dh3wr, 2017-06-28:
Ich bekomme zunehmend Rückmeldungen z.B. aus Schleswig-Holstein, wo sich ein wahres POCSAG-Sender-Paradis aufgebaut hat, dass sich die Sender gegenseitig überlagern. Vielleicht kannst noch nochmal hierrauf den Fokus legen, bitte?

7h0ma5, 2017-07-26:
Das Problem betraf nur die Transmitter vom Typ `Audio`. Der Scheduler ist jetzt überarbeitet und das Problem sollte nicht mehr auftreten. Ich habe ein neues Repository mit dem Namen `unipager-testing` eingerichtet, in dem immer die neusten Versionen vom master-Branch landen, so können wir das erstmal auf wenigen Instanzen testen. Man muss dafür einfach in der Datei `/etc/apt/sources.list.d/unipager.list` den Namen des Repositories von `unipager` in `unipager-testing` ändern und dann `apt-get update` und `apt-get upgrade` ausführen.

`unipager.list`:

    deb     http://www.afu.rwth-aachen.de/debian unipager-testing main
    deb-src http://www.afu.rwth-aachen.de/debian unipager-testing main

dh3wr, 2017-11-20:
Ich habe das Fehlverhalten nicht mehr beobachtet. Kann geschlossen werden und ein neues Stable-Release gemacht werden.

closed 2017-11-26 by 7h0ma5