rwth-afu/UniPager #71: Logausgabe in schön für Display

dh3wr, 2017-04-24:
Neue Log-Ausgabe mit max. 80 Zeichen als eigenen JSON-Inhalt ausgeben

dh3wr, 2017-09-05:
Wenn man sich mal 
``{"Log":[3,"Received Message { id: 69, mtype: AlphaNum, speed: Baud(1200), addr: 1004, func: AlphaNum, data: \"21245.0 XR1SDC von ON3DMX um 1929z\" }"]}`` vornimmt, ist da ja schon fast JSON, nur ``speed: Baud(1200)`` müsste vielleicht ein Array mit [Einheit,Speed] und [String,Number] sein?

Und für mtype und func vielleicht auch Strings? Dann könnte man das für das Display schon parsen.
Oder ist das valides JSON und ich bin nur zu dumm?

7h0ma5, 2017-09-05:
Es gibt jetzt ein eigenes Format für Messages, unabhängig vom Log:
```json
{"Message":{"id":0,"mtype":"AlphaNum","speed":{"Baud":1200},"addr":67544,"func":"AlphaNum","data":"Test"}}
```

7h0ma5, 2017-09-05:
Wird leider gerade für ARM nicht gebaut, weil der ODROID offline ist? http://ci.db0sda.ampr.org:8080/computer/ODROID/

closed 2017-09-05 by 7h0ma5

dh3wr, 2017-09-06:
Ist der ODROID der ci.db0sda.ampr.org? Denn der ist erreichbar; der Link geht ja. Läuft der Jenkins Client auch auf ci.db0sda? Dann müsstest du vielleicht den Client neu starten?

7h0ma5, 2017-09-06:
ci.db0sda.ampr.org ist die virtuelle Maschine, auf der Jenkins läuft. Der ODROID ist auf 44.225.164.14 und von mir nicht erreichbar.

dh3wr, 2017-09-06:
Aha, den hatten wir gar nicht in der hamnetdb aufgenommen. Ich füge das eben ein und starte mal den ODRIOD neu im Keller.

dh3wr, 2017-09-06:
Der .14 ist wieder online. Der Lüfter war stehengeblieben. Übrigens:
``[09/06/17 15:52:43] [SSH] WARNING: SSH Host Keys are not being verified. Man-in-the-middle attacks may be possible against this connection.`` Vielleicht kann man dagegen was tun? Wenn auch low-priority.

Es scheint, dass jetzt die Release gebaut wird.

dh3wr, 2017-09-06:
Es gibt auch ein paar Aktualisierungen für Jenkins, falls du mal Lust hast...

commit bbc37daa (2017-09-05): Forward received messages via WebSocket (close #71)