rwth-afu/UniPager #73: Nach Update auf 0.5.1 geht Unipager nicht mehr

dj7lc, 2017-05-13:
Ich habe soeben das Update auf 0.5.1 durchgeführt:
sudo apt update
sudo apt upgrade
Unipager wurde als zu aktualisierendes Paket gelistet und entsprechend aktualisiert.
Unipager wird jetzt nicht mehr automatisch gestartet und lässt sich auch manuell mit
sudo systemctl start unipager.service
oder
sudo systemctl start unipager
nicht mehr starten.
systemctl gibt aus
● unipager.service                             loaded failed failed    UniPager POCSAG transmitter controller
Unter top werden die Unipager-Prozesse ebenfalls nicht gelistet.

7h0ma5, 2017-05-15:
Nachdem ich gestern stundenlang nach der Ursache gesucht habe, sollte es mit der neusten Version wieder gehen. Das Problem war, dass die Standardbibliothek seit neustem für die ARMv7 Architektur kompiliert wird, der Raspberry Pi 1 unterstützt aber nur ARMv6-Instruktionen. Jetzt wird explizit immer die Standardbibliothek für ARMv6 verwendet.

closed 2017-05-15 by 7h0ma5

dj7lc, 2017-05-15:
Der Sender geht wieder! Ich weiß aus eigener Erfahrung, wie es ist nach solchen Fehlern zu suchen. Danke!