rwth-afu/UniPager #78: Testbutton instant

dh3wr, 2017-06-20:
Die Test-Funktion ist ja nun ganz schön, aber für den Skyper-Abgleich muss man das Signal wechselnd ein- und ausschalten. Das ist bei der aktuellen Implementierung nicht so einfach möglich. Kann man nicht einfach den Scheduler stoppen und dann mit einem Button ein 10 Sek Wave-File abspielen? Oder die Sendezeit im Testmode einstellbar machen?

DL5SMB, 2017-06-20:
Ich habe vor 2-3 Wochen einen  Skyper umgebaut und mich beim anschließenden Abgleich mit der Funktion ehrlich gesagt auch relativ schwer getan. Mir hat die vorgebene Zeit, bis der  Unipager wieder zurück in den Schedule springt einfach nicht gereicht. 

Mit  Erfahrung geht das sicherlich schnell, mir aber hätte ein längeres Testintervall (30 Sekunden oder eine Minute) oder gar ein Schalter Testmode - Schedule sichtlich geholfen.  Ich bin leider nur Amateur, kein E-Technik Ingenieur :-)

Mein Rechner steht gegenüber vom Basteltisch und da ist die Zeit wirklich knapp : 

Programm starten, aufstehen rübersetzen um dann mit Signal  erst einmal das Oszi richtig einzustellen. Hatte ich  lange  nicht mehr gemacht, allein bis ich dann alles richtig eingestellt hatte, musste ich schon 3-4 mal mindestens neu starten.  

Und mit der Kalibrierung gings dann erst richtig los. Mit Erdungsband anlegen, Schwelle links und rechts suchen, nicht finden, dann doch finden, kapazitativ nicht neutralem Abgleichbesteck usw. usw.  habe ich wirklich sehr oft neu gestartet (und unter anderem den Raspi auch 3 mal komplett zum Aufhängen gebracht - das permanente Testmode-Schedule hat er irgendwie nicht gemocht) , bin sehr viel gelaufen und war auch sehr lange beschäftigt. 

Eine Verbesserung fände ich gut, der nächste Pager wartet schon.

Wie angeführt, für Profis sicherlich alles gut machbar. Aber wenn man das nicht regelmäßig macht, ist es echt Hardcore.

dh3wr, 2017-12-08:
Bzw. wie gestern besprochen sowohl die Zeit angebbar, als auch die Frage 0101 oder random ASCII.
Und evlt. eine Interval-Lösung.