rwth-afu/UniPager #79: Aktuelle Release kaputt?

dh3wr, 2017-06-21:
Mike schreibt:
Hallo Ralf,
Seit heute morgen läuft der Unipager bei mir nicht mehr.
Auch nachdem ich das Raspbian neu auf die karte kopiert habe lässt es sich nicht mehr starten.
Gab es ein Update? Ist das fehlerhaft?
Habe aus der Debian quelle instaliert.

Konsole schreibt folgendes:
````
pi@raspberrypi:~ $ sudo unipager
UniPager 0.6.0
Copyright (c) 2017 RWTH Amateurfunkgruppe

This program comes with ABSOLUTELY NO WARRANTY.
This is free software, and you are welcome to redistribute
and modify it under the conditions of the GNU GPL v3 or later.
<https://www.gnu.org/licenses/gpl-3.0.txt>

thread '<unnamed>' panicked at 'called `Result::unwrap()` on an `Err` value: Error { repr: Os { code: 98, message: "Address already in use" } }', /checkout/src/libcore/result.rs:860
note: Run with `RUST_BACKTRACE=1` for a backtrace.
thread '<unnamed>' panicked at 'called `Result::unwrap()` on an `Err` value: WS Error <Io(Error { repr: Os { code: 98, message: "Address already in use" } })>', /checkout/src/libcore/result.rs:860
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: RecvError', /checkout/src/libcore/result.rs:860
WARN - Using dummy transmitter
thread '<unnamed>' panicked at 'called `Result::unwrap()` on an `Err` value: WS Error <Queue(Disconnected)>', /checkout/src/libcore/result.rs:860
````
Kannst du mir sagen was da jetzt los ist?

Vy 73 de DO3LK

dh3wr, 2017-06-21:
Der UniPager läuft aber nicht schon durch systemd? Mach mal ps ax | grep unipager

7h0ma5, 2017-06-21:
Der UniPager läuft definitiv schon.

dh3wr, 2017-06-21:
ok, ich habe ihm das auch geschrieben. Dann machen wir das wieder zu.

dh3wr, 2017-06-21:
vielleicht baust du da mal ein Lockfile oder so was in der Art ein gege Doppelstart?

closed 2017-06-21 by dh3wr

dj7lc, 2017-06-21:
Welchen meinst Du denn? DM0MAX ist da und sonst sehe ich keinen Sender, der auf Deinen Namen eingetragen ist.

dj7lc, 2017-06-21:
DO3LK ist im Webinterface "ONLINE", von daher gehe ich mal davon aus, dass es läuft.