rwth-afu/UniPager #81: Implement C9000 PA-Dummy Config

dh3wr, 2017-07-10:
Schon mit @phl0 besprochen:

Es gibt nun eine OpenSource Variante des PA-Dummies auf Arduino Nano Basis. Dieser kann über den serial-Port durch Senden eines Byte so eingestellt werden, dass er die dem Byte entsprechende Sendeleistung der durch ihn emulierten PA ausgibt. Dazu wäre schön:

- In der C9000 Config die Angabe eines Serial Ports (default wäre ``/dev/ttyACM0``)
- Ein drop-Down-Feld mit entsprechenden Sendeleistungen. Die genauen Byte-Werte und entsprechenden Sendeleistungen muss ich noch herrausfinden. Werden hier später kommuniziert
- Bei der Transmitter-Init wird der angegebene Port aufgemacht und das entsprechende Byte an den Arduino ausgegeben. Danach kann der Port wieder zugemacht werden.
- Ein Klick auf ``Save Config`` soll auch diesen Prozess auslösen (ist vielleicht schon impliziert enthalten)

phl0, 2017-08-01:
Ich hab mal angefangen:

[image]

So in der Richtung?

https://github.com/phl0/UniPager/tree/dummyConfig

7h0ma5, 2017-08-01:
Danke, sieht soweit sehr gut aus! Da nicht alle C9000 mit diesem Dummy ausgestattet sind, müsste es noch eine Option geben, um das Ganze zu aktivieren/deaktivieren. Dafür würde ein `bool` in der `C9000Config` reichen.

phl0, 2017-08-01:
Ich hab meinen branch mal aktualisiert. Sieht jetzt so aus:

[image]

Ein bool für die Aktivierung ist drin. Fehlt noch der Code für das schreiben auf das Serial Interface. 

@dh3wr Wie sind die Parameter für das Serial Interface? 9k6 8n1? Und was für Werte brauchts da zu schreiben. Ich hab als Werte erst mal die aus der Tabelle von http://akafunk.faveve.uni-stuttgart.de/pocsag/compact_dummy.html genommen.

phl0, 2017-08-01:
Ich hab ein Update eingecheckt. Ich kann zumindest mal auf das Serial interface schreiben. Was mir noch nicht ganz klar ist, ist wie das Byte aussehen muss. 

@7h0ma5 Ist der Code soweit grundsätzlich erst mal brauchbar?

7h0ma5, 2017-08-02:
Der Code sieht gut aus. Das Setzen der Leistung sollte bei der Initialisierung des Transmitters ausgeführt werden, also in der Funktion `C9000Transmitter::new` (`src/transmitter/c9000/transmitter.rs`), und nicht nur beim Abspeichern der Konfiguration. Wenn die Konfiguration geändert wird, wir der Transmitter automatisch neu initialisiert.

phl0, 2017-08-02:
Danke für den Tip. Ich verschiebe den Code-Teil.

phl0, 2017-08-02:
Ein Update, das die Ausgabe entsprechend der Info von @dh3wr anpasst. Es wird jeweils ein Byte ausgegeben für 16 mögliche Werte des PA-Dummy (0x00 bis 0x0F).

[image]

Im Dropdown-Menü des C9000 im UniPager werden jeweils die Ausgangsleistung am Steuersender und in Klammern dahinter die durch den PA-Dummy vorgegaukelte Leistung der Endstufe angezeigt.

phl0, 2017-08-02:
Mit https://github.com/phl0/UniPager/commit/c05705b0d136dfd6d085a42e5e24ae915a9c0059 hab ich das Webif noch mal ein wenig hübsch gemacht.

@dh3wr Könntest du diesen Stand mal mit einem PA Dummy testen? Ich hab grad weder diesen noch einen C9000 zur Hand ...

phl0, 2017-08-08:
Ich hab in Kooperation mit @dh3wr den aktuell funktioniereden Stand mal PRed mit https://github.com/rwth-afu/UniPager/pull/87.

closed 2017-08-10 by 7h0ma5