rwth-afu/UniPager #82: Zeitaussendung

dk4pa, 2017-07-15:
Hallo, gäbe es die Möglichkeit eine Zeitaussendug manuell auszusenden.
Mein Problem, wenn ich nicht mit dem Master verbunden bin kommen natürlich keine Zeitaussendungen, die der Skyper aber einmalig braucht um Meldungen zu empfangen. Grade nach dem Abgleich würde ich gerne Testen ob der Pager läuft. Leider verschwand mein angelegter Sender, sodass ich nur offline arbeiten konnte.
Vy73
Stefan

dj7lc, 2017-07-15:
Guten Abend Stefan,
die Zeitaussendung kommt jede Minute.
Dein Sender scheint tatsächlich weg zu sein, ich lege den mal neu an. Die Zugangsaten kommen gleich per E-Mail.
Vy73 Johannes

dk4pa, 2017-07-15:
Hallo Johannes, korrekt aber nur wenn man mit dem Master verbunden ist.
Offline testen geht so nicht. War der Sender jetzt wieder weg? Ralf hatte ihn mir gestern wieder eingerichtet.
Deswegen wäre die Idee gewesen, die Zeit manuell aussenden zu können, ohne das ich mich mit einem Master verbinden muss.

dj7lc, 2017-07-15:
Keine Ahnung was da los ist. Momentan bekomme ich immer 500er Error und ob das passiert was ich geklickt habe passiert ist Glücksspiel.
Wenn ich Dich richtig verstehe, möchtest Du einen Button im Webinterface haben, der eine Zeitaussendung macht. Das wäre in der Tat praktisch.

dk4pa, 2017-07-15:
Das wäre mein Vorschlag gewesen. Ich nutze zum Testen / Abgleichen Unipager, auch offline. 
Bzw. ohne Senderdaten was natürlich immer ERROR erzeugt, da Unipager sich ja versucht zu verbinden.
Hilfreich wäre vielleicht auch hier eine Hand/Auto Funktion.

dh3wr, 2017-12-28:
Man könnte auch die Zeitnachricht aus dem Unipager heraus manuell aussendbar machen. Also einen Button ``Send time``. Das Format ist ja bekannt.

7h0ma5, 2018-03-27:
Da man jetzt bei den Testnachrichten auch Typ und Function Bits auswählen kann, lässt sich die Zeit auch manuell mit dem Typ `Numeric` und Function `0` im Format `HHMMSS DDMMYY` aussenden. Meiner Meinung nach lohnt sich dafür kein eigener Button, das Webinterface ist auch so mittlerweile schon ziemlich voll.

closed 2018-03-27 by 7h0ma5

DM2NT, 2020-01-05:
Geht leider nicht. Da die 3 Leerzeichen nicht genommen werden und immer nur eins gesendet wird ;-(