rwth-afu/UniPager #83: Does UniPager support arm64 ?

tokio69, 2017-07-19:
When i want to install on a Pi with a H5 CPU from Allwinner I get this error : 
N: Skipping acquire of configured file 'main/binary-arm64/Packages' as repository 'http://www.afu.rwth-aachen.de/debian unipager InRelease' doesn't support architecture 'arm64'

7h0ma5, 2017-07-19:
Unfortunately we don't have the hardware to provide precompiled packages for ARM64. But I don't see a reason why UniPager shouldn't work on ARM64 if you compile it yourself. You can find the instructions here: https://github.com/rwth-afu/UniPager#compilation

tokio69, 2017-07-19:
Thank you for the fast reply. All went fine on a rasp but this new Orange Pi with H5 cpu is killing me. 
Compling fails also, I will have to dismiss the H5 for now I guess. Also the automated curl as described failed on installing UniPager.
You may close this issue if you like

error: Could not compile `syn`.
warning: build failed, waiting for other jobs to finish...
error: Could not compile `serde_json`.
warning: build failed, waiting for other jobs to finish...
error: build failed

dh3wr, 2017-07-19:
We can buy one and try. If it works, we can also provide an package for ARM64. What do you think @7h0ma5

tokio69, 2017-08-21:
Any news on this one? I have 3 boards with the H5 CPU and cannot do anything about it for now. So I ordered another 3 boards with the H3 cpu. That shouild work for now. I still have a SSL error at this stage when installing rustpager, but hope that will solve itself soon.