rwth-afu/UniPager #85: Problem with C9000 and UART

dh3wr, 2017-07-29:
Ich habe sowohl enable_uart=1 als auch Bluetooth aus auf einem Pi3.
Dennoch
````
UniPager 0.6.0
Copyright (c) 2017 RWTH Amateurfunkgruppe

This program comes with ABSOLUTELY NO WARRANTY.
This is free software, and you are welcome to redistribute
and modify it under the conditions of the GNU GPL v3 or later.
<https://www.gnu.org/licenses/gpl-3.0.txt>

INFO - Trying to connect to master...
INFO - Initializing C9000 transmitter...
INFO - Detected Raspberry Pi 3 Model B
thread '<unnamed>' panicked at 'Unable to open serial port: Error { kind: Io(Other), description: "Input/output error" }', /checkout/src/libcore/result.rs:860
note: Run with `RUST_BACKTRACE=1` for a backtrace.
INFO - Connection established.
INFO - Set TimeSlots { 0123456789ABCDEF }
````

Interessanterweise geht aber minicom auf /dev/ttyAMA0:
``sudo minicom -D /dev/ttyAMA0 -b38400``
Wenn ich da tippe, leuchten auch die LED auf der C9000-Platine. Der Port scheint also grundsätzlich benutzbar zu sein, nur UniPager meint das Gegenteil.

Wenn du mal schauen willst: pi@44.225.164.235, PW wie immer.

dh3wr, 2017-07-29:
Vielleicht geht in transmitter.rs das hier schief?
````
       let model = Model::get();
        info!("Detected {}", model);

        let mut serial = serial::open(model.serial_port()).expect(
            "Unable to open serial port"
);
````

7h0ma5, 2017-07-30:
Für den Raspberry Pi 3 Model B wird `/dev/ttyS0` als serieller Port genommen:

 ```rust
impl Model {
...
    pub fn serial_port(&self) -> &'static str {
        match self {
            &Model::V3B => "/dev/ttyS0",
            _ => "/dev/ttyAMA0"
        }
    }
}
```

Das habe ich wohl damals irgendwo gelesen, dass das so sein soll. Scheint aber doch nicht so zu sein, obwohl `/dev/ttyS0` existiert.

closed 2017-07-30 by 7h0ma5

7h0ma5, 2017-07-30:
Die neue Version ist jetzt im `unipager-testing` Repository verfügbar.

dh3wr, 2017-07-30:
Danke für den schnellen fix. Geht und kann in den Master branch übernommen werden.

commit fe1d94da (2017-07-30): Change serial port for Pi 3 Model B to /dev/ttyAMA0 (close #85)