rwth-afu/UniPager #88: RasPi B wird nicht erkannt

dh3wr, 2017-08-11:
Tino DM2NT schreibt:
``Raspberry ist ein B1, also ein ätwas älteres Modell. In Unipager kommt auch immer die Meldung "unknown Raspberry detected" oder so ähnlich.``

7h0ma5, 2017-08-11:
Dann brauche ich bitte den Inhalt von `/proc/cpuinfo`.

dh3wr, 2017-08-11:
Der Inhalt:
````
processor       : 0
model name      : ARMv6-compatible processor rev 7 (v6l)
BogoMIPS        : 897.37
Features        : half thumb fastmult vfp edsp java tls
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xb76
CPU revision    : 7
Hardware        : BCM2835
Revision        : 100000d
Serial          : 000000009d6ace84
````

7h0ma5, 2017-08-11:
Der Raspberry Pi wird nicht erkannt, weil der anscheinend übertaktet ist, dann wird laut [elinux.org](http://elinux.org/RPi_HardwareHistory#Board_Revision_History) auf die Revision 0x10000000 drauf addiert. Merkwürdigerweise sind das in diesem Fall aber nur 0x1000000. Ich habe das jetzt erstmal so gelöst, dass nur die hinteren 24 Bit der Revision berücksichtigt werden, da bisher kein Raspberry Pi eine höhere Revisionsnummer hat. Eine Testversion müsste gleich automatisch im testing-Repo auftauchen.

commit 522b5902 (2017-08-11): Ignore first 8 bit of the RaspberryPi revision (fix #88)
closed 2017-08-11 by 7h0ma5