rwth-afu/UniPager #90: RasPagerV1 und RasPi3 gehen nicht

dh3wr, 2017-08-31:
Laut Stefan M. und DF6EF geht das RasPagerV1 nicht mit einem RasPi3. Vielleicht schauen wir heute abend mal danach?

dh3wr, 2017-09-04:
Neue Info von Stefan M.:
``Über eine Info, falls Sie eine Lösung finden würde ich mich freuen... werde mich am Wochenende auch nochmal dran setzen, wir hatten schon mal ein ähnliches Problem, dies hatte wohl mit dem timing zu tun, dort lag die Abhilfe in der Veränderung der Prozessorfrequenz 
``

DO4DY, 2017-10-18:
Ich habe bis jetzt auch das problem das es auf einem pi3 nicht laufen möchte. Software ist da und funktioniert aber es findet keine Aussendung statt.

dj7lc, 2017-10-18:
Ich habe genau das Problem wie DO4DY mit einem RasPi 1B und Raspbian Stretch. Liegt das möglicherweise am Betriebssystem?

DO4DY, 2017-10-18:
scheinbar liegt es bei mir an einem leistungs problem es kommen nur -75dbm am ausgang anund diese wurden am frequenzzähler nicht mehr dargestellt am Spectrum Analyzer alerdings nach ein wenig einstellen schon

dh3wr, 2017-10-19:
@dj7lc Kannst du die Software mal per ``sudo systemctl stop unipager`` stoppen uns als root aus ``/var/lib/unipager`` als Working Dir starten?

Also
````
cd /var/lib/unipager
sudo /usr/bin/unipager
````
Dann kann man vielleicht auf der Konsole noch mehr sehen, was der Grund sein könnte?

DO4DY, 2017-10-22:
So bei meinem Problem lagen die 3,3 Volt nicht am HF Aussgang an deswegen hatte ich auch keine Leistung

dk4pa, 2017-11-06:
Hallo, geht es um das Profil RasPagerV1 über UniPager oder eine Hardware Version?
Vy73

dh3wr, 2018-02-27:
Bei mir geht die Version 1.0.0 mit RaspagerV1 und Raspi 3B

dh3wr, 2018-02-27:
Hat sich dann erledigt?

rtausend, 2018-02-27:
Läuft damit wieder, super! Vielen Dank

closed 2018-03-17 by 7h0ma5