rwth-afu/UniPager #92: 512 Baud possible

dh3wr, 2017-09-22:
Received Email:
````Am 22.09.2017 um 06:14 schrieb Kody Moore:
Is there a way to get unipager to transmit 512  baud?
Thanks
````

robo-w, 2020-11-22:
@dh3wr : The current version of UniPager shows a selection for 512, 1200 and 2400 baud. On a Raspberry Pi with "Audio" output selected the generated output signal is always the same for all baud rates (verified on oscilloscope). Is there a limitation to specific devices for the baud rate selection?

robo-w, 2020-12-01:
Code analysis showed, that 1200 baud is hardcoded while generating the message. The linked pull request introduces a variable baudrate at least for the transmitter of type "Audio".