rwth-afu/UniPager #93: Round-Robin der Nodes

dh3wr, 2017-09-26:
Man kann ja zur Zeit genau einen DAPNET-Core eintragen. Wie wäre es denn, wenn sich der UniPager die Nodes-Liste inkl. deren IPs holt und dann Round Robin macht? Damit schwenken die Sender bei Ausfall eines Cores direkt um. Allerdings geht das natürlich nur bei nativen Hamnet-Cores und Hamnet-Sendern. Vielleicht würde man auch einen "Home"-Core mit Bevorzugung markierbar machen? Und das ganze natürlich nur bei Bedarf aktivieren.

dh3wr, 2017-11-21:
Hier mal eine Übersicht der Node-Adressen, die man durchprobieren kann:

dapnet.afu.rwth-aachen.de
dapnet.db0sda.ampr.org
dapnet.db0fa.ampr.org
debian2.dl4ste.ampr.org
dl5ml.db0sda.ampr.org
db0rta.ampr.org
dapnet.db0vvs.de.ampr.org
db0wa.ampr.org
dapnet.ampr.org
on3dhc.db0sda.ampr.org

Man könnte die manuell als Start-Punk pflege und bei Update ggf. ein neues Release rausbringen. Der Algorithmus könnte dann sich beim gerade verbundenen DAPNET-Node per API die aktuelle Node-Liste besorgen mit einem Read-Only-Accout und entsprechend seine interne Datenbank updaten. Damit ist man auch vor dynamischen Änderungen sicher, solange zumindest eine dieser oben genannten Nodes erreicht werden kann.

In Anbetracht des Ausfalls vor 2 Wochen und meiner Abwesenheit würde die Umsetzung die Stabilität des Netzes stark verbessern. Vielleicht kommst du ja dazu?

dh3wr, 2017-12-08:
Kann man die Server irgendwo sinnvoll als Text oder Link auf der Seite aufführen, so dass es für den Nutzer transparenter ist?

dh3wr, 2017-12-10:
Feedback: geht super, Danke!
````
info Trying to connect to master...

error Connection failed.

Disconnected from UniPager.

Connected to UniPager.

warn Connecting to next fallback server...

info Connection to dapnet.db0sda.ampr.org:43434...

info Connection established.

info Set TimeSlots { 08 }

info Received Message { id: 0, mtype: AlphaNum, speed: Baud(1200), addr: 39573, func: AlphaNum, data: "NAGIOS RECOVERY: raspager.db0eif.ampr.org UniPager Connection to DAPNET Core OK" }

info Received Message { id: 1, mtype: AlphaNum, speed: Baud(1200), addr: 48159, func: AlphaNum, data: "NAGIOS RECOVERY: raspager.db0eif.ampr.org UniPager Connection to DAPNET Core OK" } 
````

dh3wr, 2017-12-11:
Mal ein Release bauen? Da ist ja wirklich ein sinnvolles Feature...

dh3wr, 2018-01-12:
Kann man den aktuell verbundenen Node auch dynamisch anzeigen?

7h0ma5, 2018-02-10:
Mit der neusten Version wird der aktuell verbundene Master im Webinterface angezeigt. Außerdem kann man jetzt unter `http://[ip]/status` ein paar Statistiken abrufen, ich benutze zum Beispiel [Grafana](https://grafana.com/) und [Telegraf](https://www.influxdata.com/time-series-platform/telegraf/) um damit einen Graphen zu generieren. Beispiel der Daten:

```json
{
  "connected": true,
  "transmitting": false,
  "timeslots": [false,false,true,true,true,true,false,true,false,false,true,true,true,true,false,true],
  "timeslot": 2,
  "queue": 15,
  "master": "dapnet.afu.rwth-aachen.de",
  "version": "1.0.0",
  "calls_tx": 126,
  "calls_rx": 141
}
```

[image]

commit 854eb895 (2017-11-26): Implement fallback server list (close #93)
closed 2018-02-10 by 7h0ma5