rwth-afu/UniPager #95: Sendung startet nicht

DL5BCA, 2017-11-10:
ich habe auf einem neuem Raspberry 3b (OS: Raspian Stretch 2017-09-07) den UniPager gemäß GitHub – Anleitung installiert.

Ein GM1200 und eine PTT-Schaltung mit BC547 und Elko für NF ist gebaut und funktioniert.

Ich kann die UniPager Oberfläche im Browser aufrufen.
Zur Zeit ist noch kein Rufzeichen und Auth Key eingetragen, da  sich Ralf noch in Thailand befindet.
Ich habe PTT Config auf PTT Method: GPIO und GPOI Pin auf 13 eingestellt.
Leider tastet das UniPager Programm die PTT nicht. 

Vom Terminalfenster lässt sich mit  „gpio export 13 out“ und „gpio -g write 13 1“ bzw „gpio -g write 13 0“ ein- und ausschalten.

Hat jemand eine Idee?

dk4pa, 2017-11-10:
Hallo, 
meinst du HeaderPin 13? Das wäre dann GPIO_GEN2 also in UniPager 2.

Ich nutze GPIO 0, das ist HeaderPin 11.

Gruß
Stefan

DL5BCA, 2017-11-10:
Hallo Stefan,
danke für die Antwort.
Genau das war es. Jetzt klappt es...

73 Thomas DL5BCA

dk4pa, 2017-11-10:
Gerne.
Dann kann hier zugemacht werden ;)

DL5BCA, 2017-11-15:
Leave a comment

closed 2017-11-15 by DL5BCA