rwth-afu/UniPager #96: C9000: error Unable to write data to the serial port

dh3wr, 2017-12-04:
Dieses kommt nun bei jeder Sendung, wahrscheinlich bei jedem SerWrite auf die UART. Version 0.7.0.
Aufgetreten bei c9000.db0wa.ampr.org

Der Port ist aber noch da:
````
pi@c9000:/dev $ ls -l /dev/ttyAMA0
crw-rw---- 1 root dialout 204, 64 Sep 28 21:22 /dev/ttyAMA0
````
Soll ich etwas noch testen oder überprüfen, um den Grund besser herausfinden zu können?

dh3wr, 2017-12-05:
sudo systemctl restart unipager hat keine Änderung gebracht. Nun sudo reboot

dh3wr, 2017-12-05:
Das hat es behoben...

7h0ma5, 2017-12-05:
Kann ich mir nicht erklären. Hattest du versucht den Port mit einem anderen Programm zu öffnen und zu beschreiben als das Problem noch bestand?

dh3wr, 2017-12-05:
Nein, leider nicht. Aber wäre eine gute Idee gewesen. Ich versuche mal in Nagios-Plugin zu basteln, was die Log-Ausgabe ``error Unable to write data to the serial port`` erkennt und sich dann meldet. Damit wir das in Zukunft schneller mitbekommen. Ebenfalls erhöhe ich den Zeitraum für die ``/var/log/syslog``, damit wir evtl. dort einen Hinweis finden.

dh3wr, 2017-12-05:
Nagios-Monitoring ist aktiv, wenn auch etwas brute-force:
[image]
Könnte man eine bessere Status-Abfrage implementieren? Der ``Connected`` Status macht ja den Anschein, dass alles ok ist. Dann könnte man das auch einfacher monitoren. Z.B. Anzahl der Fehler in der letzten Stunde.

dh3wr, 2017-12-28:
Wir tauschen heute den Sender bei DB0WA aus, dann kann man sich die Sache hier auf dem Labortisch nochmal anschauen.

dh3wr, 2017-12-29:
Grund gefunden: Es hat sich der Bluetooth-Stack (vielleicht durch ein Update) aktiviert. Dadruch war /dev/ttyAMA0 verbogen.

closed 2017-12-29 by dh3wr