rwth-afu/UniPager #98: Enhancement: Auswahl Unteradressen und Nachrichtentyp

dk4pa, 2017-12-08:
Mit der Möglichkeit die Skyper zu programmieren, lassen sich für eine RIC, oder eine Zweite ( je nach Skypertyp) auch nur bestimme Unteradressen ansteuern.

Auch bei anderen Pagertypen ist es ja möglich, Unteradressen einer RIC vielfältig zu nutzen.

Ebenfalls wäre es gut wenn man über das Senderinterface auswählen kann ob eine Nachricht als numerisch oder alphanumerisch gesendet werden soll.

Als Standard könnte ja immer alphanumerisch und Unteradresse D stehen. 
Die Möglichkeit wäre hier dann verschiedene Meldertypen anzusteuern.
Gerade für Tests mit „Over Air Programm“ für Zeit senden wäre dies von Vorteil und um eine RIC ggf. mehrfach auszunutzen.

Vy73

closed 2017-12-26 by 7h0ma5

dk4pa, 2017-12-28:
Hi,

leider funktioniert es noch nicht so wie gedacht.

Speed kann zwar geändert werden, wird auch in UniPager in der Aussendung so angezeigt, jedoch wird das Signal mit 1200bd gesendet.

Wenn MessageTyp = AlphaNum und MessageFunc ist 0 oder 1 werden die Rufe BCD als Numeric kodiert.
bei MessageFunc 2 und 3 werden die Rufe AlphaNum kodiert ( 7-bit ASCII )

Die Funktionen müssten eigentlich so sein, dass überprüft wird, wenn MessageTyp = Alphanum
dann bei MessageFunc 0-3 als 7-bit ASCII Alphanum kodiert wird.

Kann es hiermit zusammenhängen?
https://github.com/rwth-afu/UniPager/blob/d44b2fcae4cad9807fb86b92e5a7b9bed87346f4/src/pocsag/generator.rs#L116

vy73
Stefan, dk4pa

7h0ma5, 2017-12-30:
Hallo Stefan,

die Kodierung wurde tatsächlich bisher anhand der Funktionsbits gewählt, was natürlich nur für eine begrenzte Art von Pagern sinnvoll ist. Ich habe das jetzt geändert, dass die Kodierung nur vom MessageType abhängt.

Die Auswahl der Baudrate hat noch keine Wirkung, da es bisher keinen Sender gibt, der andere Baudraten unterstützt. Für den Audio-Sendertyp wäre das aber relativ einfach zu implementieren. Allerdings kommt mit verschiedenen Baudraten noch ein anderes Problem dazu: Nachrichten mit verschiedenen Baudraten können nicht gemischt werden, das ganze Preamble muss dann auch in der anderen Baudrate gesendet werden. Das heißt man benötigt für jede Baudrate eine eigene Warteschlange. Das gibt dann auch wieder neue Probleme bei der Priorisierung (siehe auch https://github.com/DecentralizedAmateurPagingNetwork/Core/issues/122).

73, Thomas DL2IC

dk4pa, 2017-12-30:
Hallo Thomas,

besten Dank, werde es später direkt testen. 

Habe auch gesehen, dass du die Bezeichnung direkt in Func0-3 als Ausgabe in UniPager geändert hast.
Das wirkt klarer als wenn Nuneric oder Activation steht ;)

Vy73 ,Stefan dk4pa

commit 83d8a63f (2017-12-26): Add additional message options (close #98)