rwth-afu/UniPager #99: Bei Update zu 0.7 geht die Konfiguration verloren

dj7lc, 2017-12-14:
Bei Update zu 0.7 via "apt upgrade" geht die Konfiguration verloren.

dj7lc, 2017-12-14:
Und in der Infobox sind jetzt interessante Bitmuster:
` Connected to UniPager.
warn Connecting to next fallback server...
error No callsign configured.
error Connection failed.
error No callsign configured.
error Connection failed.
warn Connecting to next fallback server...
error No callsign configured.
error Connection failed.
error No callsign configured.
error Connection failed.
warn Connecting to next fallback server...
error No callsign configured.
error Connection failed.
error No callsign configured.
error Connection failed.
warn Connecting to next fallback server...
error No callsign configured.
error Connection failed.
error No callsign configured.
error Connection failed.
warn Connecting to next fallback server...
error No callsign configured.
error Connection failed.
info Config updated. Initiating restart.
info Waiting for the connection to terminate...
info Waiting for the scheduler to terminate...
info Scheduler stopped.
warn Using dummy transmitter
info Scheduler started.
info Connection to dapnet.db0vvs.ampr.org:43434...
info Connection established.
info Set TimeSlots { 3B }
info Received Message { id: 0, mtype: Numeric, speed: Baud(1200), addr: 2504, func: Numeric, data: "112600 141217" }
info 10101010101010101010101010101010
info 10101010101010101010101010101010
info 10101010101010101010101010101010
info 10101010101010101010101010101010
info 10101010101010101010101010101010
info 10101010101010101010101010101010
info 10101010101010101010101010101010
info 10101010101010101010101010101010
info 10101010101010101010101010101010
info 10101010101010101010101010101010
info 10101010101010101010101010101010
info 10101010101010101010101010101010
info 10101010101010101010101010101010
info 10101010101010101010101010101010
info 10101010101010101010101010101010
info 10101010101010101010101010101010
info 10101010101010101010101010101010
info 10101010101010101010101010101010
info 01111100110100100001010111011000
info 00000000001001110010010100110111
info 11000100001000110000010010110101
info 10000001100110011100010110101111
info 10010100001001000111010000100000
info 10011001100110011001111101000111
info 01111010100010011100000110010111
info 01111010100010011100000110010111
info 01111010100010011100000110010111
info 01111010100010011100000110010111
info 01111010100010011100000110010111
info 01111010100010011100000110010111
info 01111010100010011100000110010111
info 01111010100010011100000110010111
info 01111010100010011100000110010111
info 01111010100010011100000110010111
info 01111010100010011100000110010111
[... 82 lines cut ...]
info 10101010101010101010101010101010
info 10101010101010101010101010101010
info 10101010101010101010101010101010
info 10101010101010101010101010101010
info 10101010101010101010101010101010
info 10101010101010101010101010101010
info 10101010101010101010101010101010
info 10101010101010101010101010101010
info 10101010101010101010101010101010
info 01111100110100100001010111011000
info 00000000001001110010010100110111
info 11000100001001001000000010010101
info 10000001100110011100010110101111
info 10010100001001000111010000100000
info 10011001100110011001111101000111
info 01111010100010011100000110010111
info 01111010100010011100000110010111
info 01111010100010011100000110010111
info 01111010100010011100000110010111
info 01111010100010011100000110010111
info 01111010100010011100000110010111
info 01111010100010011100000110010111
info 01111010100010011100000110010111
info 01111010100010011100000110010111
info 01111010100010011100000110010111 `

7h0ma5, 2017-12-14:
Handelt es sich um einen C9000? Da gab es mehrere Änderungen in der Konfiguration. Wenn es nur da auftritt kann ich vielleicht noch etwas ändern, damit das Update problemlos läuft. Die Bitmuster sind normal, wenn der Dummy-Transmitter gewählt ist, die Rohdaten werden dann zu Debugging-Zwecken ausgegeben.

dj7lc, 2017-12-14:
Ah ja, dann ist das Rätsel schon mal gelöst. Der Sender ist ein Raspager V1.

7h0ma5, 2017-12-14:
Das Problem ist gefunden und behoben. Version 0.7.1 wird gerade von Jenkins kompiliert und müsste in etwa 20 Minuten im Repository verfügbar sein. Hier findet man den aktuellen Status: http://ci.db0sda.ampr.org:8080/job/unipager-binaries/

closed 2017-12-14 by 7h0ma5

dj7lc, 2017-12-14:
Super! Gleich mal installieren...

commit 4f62f12c (2017-12-14): Fix failed configuration migration (fix #99)