rwth-afu/UniPager #105: Problem: UniPager 0.7.1 Pi2 - Software stürzt bei gedrückter PTT ab.

DL8KFO, 2018-01-16:
Folgendes Problem trott auf. Nach längerer Laufzeit (1-2 Tage) scheint UniPager abzustürzen. Dabei verbleibt GPIO0 auf Low und der GM1200 sendet dauerhaft einen stummen Träger. Dennoch bleibt der Pi im Netzwerk erkennbar, kann aber über Port 8073 nicht mehr angesprochen werden. Im DAPNET wird der TX weiter als "Online" angezeigt. Kann jemand das Problem nachvollziehen?

73
  Torsten

dh3wr, 2018-02-27:
Greift irdendetwas anderes auf GPIO0 zu?

dj7lc, 2019-05-16:
Genau diesen Fehler hatte ich mit genau dieser Version auch an einigen Sendern beobachten können. In der aktuellen Version tritt das Problem nicht mehr auf. Ich wage zu behaupten, dass wir hier zu machen können.

closed 2019-05-17 by 7h0ma5