rwth-afu/UniPager #106: Implement RasPager V2.0

dh3wr, 2018-02-26:
Es wird evtl. eine verbesserte PCB Variante geben. Dabei werden die Pins auf dem GPIO-Header anders sein. Könntest Du eine Version 2.0 implementieren, die im Wesentlichen nur eine andere GPIO-Zuordnung hat?
Nicht schön aber selten hier ein Bild:
[image]

7h0ma5, 2018-02-27:
Sehe ich das richtig, dass diese Raspager-Variante dann nicht mehr mit dem Raspberry Pi 1 kompatibel sein wird?

Aus den Pin-Zuordnungen werde ich noch nicht ganz schlau. Es gibt zum Beispiel "CLK, TXCLK und CLKOUT", welche Clock davon ist für den PLL-IC? Ich vermute mal CLKOUT? So sieht die Zuordnung bisher aus (Wiring-Pi-Definition):

```rust
pub const RASPAGER1_PINS: RaspagerPins = RaspagerPins {
    le: 0,
    ce: 7,
    clk: 3,
    sdata: 2,
    muxout: 13,
    atclk: 11,
    atdata: 10,
    handshake: 5,
    ptt: 4
};

pub const RASPAGER2_PINS: RaspagerPins = RaspagerPins {
    le: 9,
    ce: 7,
    clk: 30, // ???
    sdata: 2,
    muxout: 21,
    atclk: 11,
    atdata: 10,
    handshake: 5,
    ptt: 4
};
```

dh3wr, 2018-02-27:
Zu 1: Ja, die neue Variante ist mit Pi1 nicht mehr kompatibel.
Ich habe für ATMega-ISP die gleichen Pins wie beim RasPagerC9000 verwendet (da wissen wir, dass der Reset-Pin beim PI-Boot keinen Ärger macht)

Die alte Belegung von V1 sieht so aus:
[image]

Müsste man jetzt mal vergleichen.

commit b91aa539 (2018-02-27): Implement Raspager 2 transmitter (close #106)
closed 2018-02-27 by 7h0ma5