rwth-afu/UniPager #107: Can't send Messages with Raspager V1, Test shows unknown Raspberry Version

rtausend, 2018-02-27:
Hi,

when i try "test" i get:

INFO Detected Unknown Raspberry Pi

WARN Disconnected from master.
INFO Waiting for the scheduler to terminate...
INFO Scheduler stopped.
INFO Starting test transmission.
INFO Initializing RasPager transmitter...
INFO Detected Unknown Raspberry Pi
INFO Test transmission completed. Restarting...
INFO Connection to 44.225.111.5:43434...
INFO Initializing RasPager transmitter...
INFO Detected Unknown Raspberry Pi
INFO Connection established.
INFO Set TimeSlots { 0123456789ABCDEF }

All Timeslots are Grey, not Blue, here are some Infos about the Raspi (fresh installed!)

root@raspberrypi:~# uname -a
Linux raspberrypi 4.9.59-v7+ #1047 SMP Sun Oct 29 12:19:23 GMT 2017 armv7l GNU/Linux
root@raspberrypi:~# cat /proc/version
Linux version 4.9.59-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #1047 SMP Sun Oct 29 12:19:23 GMT 2017
root@raspberrypi:~# cat /proc/device-tree/model
Raspberry Pi 2 Model B Rev 1.1

Thanks Robert

dh3wr, 2018-02-27:
Sowohl mit der release- als auch mit der testing-Version

7h0ma5, 2018-02-27:
Dann brauche ich bitte die Ausgabe von `cat /proc/cpuinfo`.

rtausend, 2018-02-27:
Du bekommst fast alles :)

`root@raspberrypi:/opt/raspager# cat /proc/cpuinfo
processor       : 0
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

processor       : 1
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

processor       : 2
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

processor       : 3
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

Hardware        : BCM2835
Revision        : a21041
Serial          : 000000003a036a1c`

7h0ma5, 2018-02-27:
Die gefixte Version wird in etwa 20 Minuten im unipager-testing Repository verfügbar sein. Hier der aktuelle Build-Status: http://ci.db0sda.ampr.org:8080/job/unipager-testing-binaries/architecture=armhf,label=armhf/29/

commit cb8abfd3 (2018-02-27): Add new Raspberry Pi revision numbers (fix #107)
closed 2018-02-27 by 7h0ma5