rwth-afu/UniPager #112: Network connection interruption is not detected

dh3wr, 2018-04-11:
Es kommt vor, das die IP-Verbindung von Unipager zum aktuell verbundenen Core abbricht, ohne dass der Unipager dies bemerkt. Wann und wie genau ist noch nicht bekannt, aber der finale Status ist, dass Unipager "Connected" meldet, aber kein Core diesen Sender als online führt. Da Unipager meint, es wäre verbunden, bleibt dieser Zustand auch unbegrenzt bestehen.

Vorschlag: Einführung eines Timeouts afu dem Unipager von 2 Minuten. Da die Uhrzeit jede Minute kommt, kann man sehr sicher von einem Fehler ausgehen, wenn 2 Minuten keine neuen Meldungen vom Core gekommen sind.

7h0ma5, 2018-04-11:
Das ist merkwürdig, das Read-Timeout ist eigentlich schon auf 125 Sekunden eingestellt. Ist das auf einen Raspberry Pi mit Standard-Kernel passiert?

dh3wr, 2018-04-11:
Ja, alles Standard.

[image]

[image]