rwth-afu/UniPager #113: Rubriken ob 96 werden auf 32 gemappt

dh3wr, 2018-04-16:
Siehe https://github.com/DecentralizedAmateurPagingNetwork/Core/issues/159

dh3wr, 2018-04-16:
Es mag hier das falsche Mapping geschehen:
https://github.com/rwth-afu/UniPager/blob/master/src/connection.rs
Zeile 211ff. ``fn handle_message(&mut self, data: &str) -> Result<AckStatus> {``

7h0ma5, 2018-04-16:
Wenn würde das hier dran liegen: https://github.com/rwth-afu/UniPager/blob/master/src/pocsag/encoding.rs#L8

Alle Bytes größer als 127 werden zu einem 0x3F, also einem "?" gemappt, da es sich um kein gültiges ASCII-Zeichen handelt. Größer geht auch gar nicht, da Alphanumerische Zeichen im POCSAG-Standard nur 7 Bit groß sind.

dh3wr, 2019-02-04:
Liegt nicht an Unipager, sondern 95 ist die max. Rubrikennummer. Fertig.

closed 2019-02-04 by dh3wr